open-power/snap #310: HLS: Cannot understand the array in job descriptor

luyong6, 2017-06-02:
In software/example/action_intersect.h, I defined intersect_job_t as: 

```C
typedef struct {
	CONTROL Control;	/*  16 bytes */
	intersect_job_t Data;
	uint8_t padding[SNAP_HLS_JOBSIZE - sizeof(intersect_job_t)];
} action_reg;
```

luyong6, 2017-06-02:
I think this is, how to say, a weakness? 
HLS is not good at data casting.

bmesnet, 2017-06-23:
@luyong6 
Do you expect we should report that to Xilinx ?

luyong6, 2017-06-23:
Yes, let me reproduce this and look at the generated vhdl.

fhaverkamp, 2017-06-28:
I think it is worthwhile reporting if Lu Yong is able to reproduce. This I would think is really not as expected. Are the values below those arrays as expected or shifted?

jsvogt, 2017-10-20:
@luyong6 Please reproduce or close

luyong6, 2017-10-20:
Let me close it for now. (And reopen it when it harms again.)
This can be workarounded, and 108bytes actually will not allow many table entries like this.

closed 2017-10-20 by luyong6