open-power/snap #315: HLS_search: wrong results with snap_search

joergkayser, 2017-06-09:
working with branch=master and action=hls_search. I am creating a small file (random or uniform data)  of n bytes and search one char. Expect n hits on uniform data and 0..n hits on random data

for size in 1 2; do 
  char=$(cat /dev/urandom|tr -dc 'a-zA-Z0-9'|fold -w 1|head -n 1)     # random char to search for
  cat /dev/urandom|tr -dc 'a-zA-Z0-9'|fold -w ${size}|head -n 1 >${size}.rnd   # random ASCII data 
  cmd='print("A" * '${size}', end="")'; python3 -c "$cmd" >${size}.uni              # uniform data 
  count=$(fgrep -o $char ${size}.rnd|wc -l);                                                      # expected chars in file
  snap_search -m2 -p${char} -i${size}.rnd -E${count} -t500 -v                          # random search
  snap_search -m2 -pA -i${size}.uni -E${size} -t500 -v                                      # uniform search
done

The test fails, as soon as it expects >0 hits

bmesnet, 2017-06-09:
Good corner case! I'll have a look to that

bmesnet, 2017-06-23:
ok this is corrected and Jörg tested it ok

closed 2017-06-23 by bmesnet