open-power/snap #317: Make Issues w/ HLS Examples

kenhill, 2017-06-09:
`make config` skips over creating the RTL files from the HLS source

boekholt, 2017-06-12:
Did the environment variable `$ACTION_ROOT` point to the resulting directory for RTL file generation?

kenhill, 2017-06-13:
No, I was setting `$ACTION_ROOT` to the hls_<action> directory. 

Although this is a user error, adding a check for correctly formatted `$ACTION_ROOT` would be useful for HLS examples. Currently the `make config` does generate a Vivado project using create_framework.tcl, but the framework project includes all the HLS generated source code for vhdl, verilog, and systemc with an incorrect top level module selected (not psl_fpga). This causes 'make image` to fail with a nondescript error.

boekholt, 2017-06-19:
@kenhill
Sorry for the delay (was on vacation last week).

We'll discuss the check for a correctly set up `$ACTION_ROOT` for HLS actions here in the team (we are aware that the whole setup for HLS actions is not nice).

I don't understand your issue regarding the incorrect top level module.
Are you talking about the case when `$ACTION_ROOT` is not set up correctly or about the case when it is set up correctly?

Trying to recreate your problem I do not see an incorrect top level module.
The top level module for the HLS generated source code is always hls_action which then is integrated into the SNAP framework via the action_wrapper entity (hierarchy `psl_fpga -> psl_accel -> action_wrapper -> hls_action`).

kenhill, 2017-06-19:
@boekholt 
No problem, Welcome back!

Here are the strings I am using on a fresh clone of the master branch:

source /data/snap_setup.sh (some quick exports for snap_settings using KU3 board)
export SDRAD_USED=TRUE
export ACTION_ROOT=/home/$USER/snap/hardware/action_examples/hls_memcopy
make -C $ACTION_ROOT
source /home/$USER/snap/hardware/snap_settings
make -C snap/hardware config 

The above commands create a Vivado project with the incorrect toplevel module set. A screenshot of the Sources tab is attached below. 

vivado /home/$USER/snap/hardware/viv_project/framework.xpr
[image]

When `ACTION_ROOT` is set incorrectly for the HLS example, 

https://github.com/open-power/snap/blob/master/hardware/setup/create_framework.tcl#L98-L99

imports all the generated RTL code for Verilog, VHDL, and SystemC. 

The extra HLS source files causes

https://github.com/open-power/snap/blob/master/hardware/setup/create_framework.tcl#L118

to not select psl_fpga as the toplevel module.

`make -C snap/hardware image` fails to build the framework.xpr project.

tail snap/hardware/logs/snap_build.log 
  IBUF => IBUF (IBUFCTRL, INBUF): 96 instances

INFO: [Common 17-83] Releasing license: Synthesis
96 Infos, 271 Warnings, 113 Critical Warnings and 0 Errors encountered.
synth_design completed successfully
synth_design: Time (s): cpu = 00:00:30 ; elapsed = 00:00:40 . Memory (MB): peak = 2206.359 ; gain = 1016.625 ; free physical = 13582 ; free virtual = 146087
INFO: [Common 17-1381] The checkpoint '/home/nimbix/snap/hardware/build/Checkpoints/synth_design.dcp' has been generated.
                                start locking PSL                                           17:30:23              
WARNING: [Vivado 12-180] No cells matched 'b'.
ERROR: [Vivado 12-2538] No cell specified, you need to specify a cell to be locked.

boekholt, 2017-06-20:
Ok, I see - I did not create the HLS action's RTL code manually and therefore did not get those HLS modules at all (when setting `$ACTION_ROOT` correctly the HLS action's RTL code gets built automatically via `make -C snap/hardware config`.

With `$ACTION_ROOT` incorrectly set one level up after manually building the HLS action's RTL code, duplicate sources are found by `make -C snap/hardware config` because (as you mentioned already) the HLS make process generates vhdl code as well as verilog code and systemc code. We generally (for HLS based actions and for HDL based actions) integrate all vhdl and verilog sources under `$ACTION_ROOT`. Therefore, the hierarchy cannot be created. Removing the duplicate sources additionally generated by the HLS build could resolve this problem. Then also an 'incorrect' `$ACTION_ROOT` would work.

luyong6, 2017-06-23:
Why not adding directions in README.md 

$ACTION_ROOT = ..../hdl_memcopy
$ACTION_ROOT = ..../hls_memcopy/**vhdl**     need explicitely assign the source directory that contains the source vhdl/verilog files!

boekholt, 2017-06-27:
I created pull request #338 which should resolve this problem.
With that change it is for instance for the HLS memcopy example allowed to set `$ACTION_ROOT` to `$SNAP_ROOT/action_examples/hls_memcopy` or to `$SNAP_ROOT/hardware/action_examples/hls_memcopy/**vhdl**`. Both versions will work regardless if the HLS action was already pre-compiled or not.

boekholt, 2017-06-30:
@kenhill can we close this?

kenhill, 2017-06-30:
The master branch still creates a Vivado project with an incorrect toplevel module selected when `ACTION_ROOT` is set to $SNAP_ROOT/.../hls_memcopy

It seems that the `add_files` line in create_framework.tcl is also including hlsMemcopy_xcku060-ffva1156-2-e/ in the search path for files

kenhill, 2017-06-30:
The `add_files` command does have a -norecurse flag but this does not help if `ACTION_ROOT` is set incorrectly by the user. Can we add the following check back to the Makefile?

https://github.com/open-power/snap/blob/fa16ba76f3138a6e954ebe10ca6e48d6ff8aa40b/hardware/Makefile#L104-L107

boekholt, 2017-06-30:
The change is meant to make the setting of `ACTION_ROOT` to `$SNAP_ROOT/hardware/action_examples/hls_memcopy` no longer incorrect. Therefore I do not want that check back in the Makefile.
I was sure that the change was sufficient - need to double check... (will do that on Monday - sorry for the delay)

boekholt, 2017-07-03:
With change #351 I made sure that psl_fpga is top level also in cases with additional links to the HDL sources for HLS actions.

@kenhill please let me know if this is an acceptable solution for your issue (and close this issue, if so).

Thanks
Sven

closed 2017-07-04 by kenhill