open-power/snap #321: Porting SNAP to more devices

michaellass, 2017-06-14:
Currently SNAP is supported for use with the Alpha Data KU3 and the Nallatech 250S. Are there plans or ongoing work on porting this to more devices such as the Alpha Data 7v3?

Would porting require a modified PSL checkpoint file for the target device or should SNAP work using the ones for the Alpha Data CAPI Design Kit as they are provided for the 7v3 and the 8K5?

ThomasFuchs, 2017-06-22:
The SNAP work flow can use the PSL checkpoints that comes with the CAPI Design Kit.  But the current available Design Kit checkpoints for 7v3 and 8k5 have two drawbacks: 1) They was build with Vivado 2015.4. 2) They hide FPGA resources that SNAP could use (for example SDRAM, NVME,..). A team inside IBM is working on PSL checkpoint updates for the Design Kit that should solve these problems.

jsvogt, 2017-06-30:
Unlikely to be fixed at the moment, as this is a completely different FPGA chip. 
Check with IBM team again in a few weeks.

jsvogt, 2018-03-01:
We added Semptian NSA121B and Alpha-Data 8K5, but decided, not to add 7V3.

closed 2018-03-01 by jsvogt