open-power/snap #330: core dump in snap_memcopy during hw_snap_card_free

joergkayser, 2017-06-22:
When running a loop of snap_memcopy commands, we fail with a core dump during
  snap_memcopy -i 3093.in -o 3093.out -v -t154660

The core dump shows 
Program terminated with signal 11, Segmentation fault.
#0  0x000000348a408213 in pthread_join () from /lib64/libpthread.so.0
(gdb) where
#0  0x000000348a408213 in pthread_join () from /lib64/libpthread.so.0
#1  0x0000000000405347 in cxl_afu_free ()
#2  0x0000000000402b30 in hw_snap_card_free (card=0x16e5c20) at snap.c:273
#3  0x00000000004022b3 in main (argc=<optimized out>, argv=<optimized out>) at snap_memcopy.c:345

branch=HLS_examples_corrections 0844244 2017-06-21 17:32:21
core file is in /afs/bb/proj/fpga/framework/jkayser/snap_hls/hardware/sim/ies/20170622_180639

Can you add more verbosity (pointers, 'AFU handle...) into the code to show, why the AFU cannot be closed anymore ?
Thanks

ibm-genwqe, 2017-06-23:
i am not able to figure out why we get a core dump. Is this from Simulation or on Real HW ?
Can you recreate ?

joergkayser, 2017-06-26:
the core dump can be recreated with fine sizes of 2521 Bytes or higher, the test runs successfully with 2520 Bytes or less, see core dumps in ~jkayser/snap_hls/hardware/sim/ies/20170626_115000

ibm-genwqe, 2017-06-27:
Seems like a HLS problem, Frank , or Bruno have to look into this on.

fhaverkamp, 2017-06-28:
I assume we might allocate less than 64 bytes or an output buffer. We should fix that quick to see if it helps. It is certainly wrong what is done there.

fhaverkamp, 2017-06-28:
https://github.com/open-power/snap/tree/HLS_memcopy_fixups contains a potential fix, which I would like to try out.

joergkayser, 2017-06-29:
with #346 a fix was merged into the master code, which fixes this issue. Therefor I close this ticket

closed 2017-06-29 by joergkayser