open-power/snap #337: Add parameters for AXI_NVMe in action_wrapper.vhd_source

luyong6, 2017-06-27:
I am trying to make a HLS example on FGT and using NVMe port. 
With USE_NVME=TRUE, and set hls_action interface like this: 
```
void hls_action(snap_membus_t *din_gmem,
		snap_membus_t *dout_gmem,
		snap_membus_t *d_ddrmem,
               snap_membus_t *d_nvmemem,
   	       action_reg *act_reg,
		action_RO_config_reg *Action_Config)
{
	// Host Memory AXI Interface
#pragma HLS INTERFACE m_axi port=din_gmem bundle=host_mem offset=slave depth=512 \
  max_read_burst_length=64  max_write_burst_length=64 
#pragma HLS INTERFACE s_axilite port=din_gmem bundle=ctrl_reg offset=0x030

#pragma HLS INTERFACE m_axi port=dout_gmem bundle=host_mem offset=slave depth=512 \
  max_read_burst_length=64  max_write_burst_length=64 
#pragma HLS INTERFACE s_axilite port=dout_gmem bundle=ctrl_reg offset=0x040

	// DDR memory Interface
#pragma HLS INTERFACE m_axi port=d_ddrmem bundle=card_mem0 offset=slave depth=512 \
  max_read_burst_length=64  max_write_burst_length=64 
#pragma HLS INTERFACE s_axilite port=d_ddrmem bundle=ctrl_reg offset=0x050
    
      // NVMe memory Interface
#pragma HLS INTERFACE m_axi port=d_nvmemem bundle=nvme offset=slave depth=512 \
  max_read_burst_length=64  max_write_burst_length=64 
#pragma HLS INTERFACE s_axilite port=d_nvmemem bundle=ctrl_reg offset=0x060

```

I meet a simulation compile fail of "ports mismatch". 

That is because action_wrapper.vhd_source doesn't have a similar paragraph for NVMe AXI port like this 
```
-- Parameters for Axi Master Bus Interface AXI_CARD_MEM0 : to DDR memory
    C_M_AXI_CARD_MEM0_ID_WIDTH       : integer   := 1;
    C_M_AXI_CARD_MEM0_ADDR_WIDTH     : integer   := 33;
    C_M_AXI_CARD_MEM0_DATA_WIDTH     : integer   := 512;
    C_M_AXI_CARD_MEM0_AWUSER_WIDTH   : integer   := 1;
    C_M_AXI_CARD_MEM0_ARUSER_WIDTH   : integer   := 1;
    C_M_AXI_CARD_MEM0_WUSER_WIDTH    : integer   := 1;
    C_M_AXI_CARD_MEM0_RUSER_WIDTH    : integer   := 1;
C_M_AXI_CARD_MEM0_BUSER_WIDTH : integer := 1;
```

luyong6, 2017-06-27:
Current action_wrapper.vhdl_source NVMe port is like this 
```   -- AXI NVME Interface                                                                              -- only for NVME_USED=TRUE
    m_axi_nvme_araddr          : OUT STD_LOGIC_VECTOR ( 31 DOWNTO 0 );                                 -- only for NVME_USED=TRUE
    m_axi_nvme_arburst         : OUT STD_LOGIC_VECTOR ( 1 DOWNTO 0 );                                  -- only for NVME_USED=TRUE
    m_axi_nvme_arcache         : OUT STD_LOGIC_VECTOR ( 3 DOWNTO 0 );                                  -- only for NVME_USED=TRUE
    m_axi_nvme_arid            : OUT STD_LOGIC_VECTOR ( 0 DOWNTO 0 );                                  -- only for NVME_USED=TRUE
    m_axi_nvme_arlen           : OUT STD_LOGIC_VECTOR ( 7 DOWNTO 0 );                                  -- only for NVME_USED=TRUE
m_axi_nvme_arlock : OUT STD_LOGIC; 
``` 

Not under control of parameters.

boekholt, 2017-06-27:
Hi @luyong6,

yes, I simply copied the action wrapper from our hdl_example and adapted what was required for HLS. At that point in time there did not exist an HLS example which used the NVMe drives.

We can add that now.
- If you want to do it yourself you may simply adapt [hardware/hdl/hls/action_wrapper.vhd_source](../../blob/master/hardware/hdl/hls/action_wrapper.vhd_source) and create a pull request for it.
- If you want me to make the necessary changes, could you please point me to the vhdl generated for your HLS example?

luyong6, 2017-06-28:
Hi  Sven, 
Let me do it myself a little bit. One question, is C_M_AXI_NVME_DATA_WIDTH=32 for NVMe port? 
Host and DDR port are 512bits data width. Why NVMe port is so narrow?

boekholt, 2017-06-28:
Hi Lu,
the AXI interface between action and NVMe host is just a control interface for register reads and writes to the 32bit registers within the NVMe host logic.

luyong6, 2017-06-28:
Another question: 
Is AXI NVME a master, right? 
AMBA AXI spec says AWLOCK and ARLOCK are (1:0)  -- 2bits. 
but from hardware/action_example/hdl_example, M_AXI_AWLOCK/ARLOCK is only 1 bit.
It also impacts psl_accel_fgt.vhd_source

boekholt, 2017-06-29:
Yes, from the  action's perspective the AXI interface to NVMe is a master.

The nvme_host block design defines the bits M_AXI_AWLOCK/ARLOCK as `std_logic_vector(0 to 0)`. That is why we defined it the same way in [hardware/action_examples/hdl_example/action_wrapper.vhd_source](../tree/master/hardware/action_examples/hdl_example/action_wrapper.vhd_source). But, those inputs to the nvme_host are not used at all.

So, if HLS requires AWLOCK and ARLOCK as (1:0) you may add the necessary conversion to [hardware/hdl/hls/action_wrapper.vhd_source](../tree/master/hardware/hdl/hls/action_wrapper.vhd_source).

**Note:** This morning we merged the branch `nvme_update` into master.

boekholt, 2017-07-06:
@luyong6
FYI: We are currently changing the action wrappers for partial reconfiguration support. This might affect the changes you are adding to the wrapper for HLS.
Are you still working on this? Please let me know if I can provide some help.

luyong6, 2017-07-06:
Hi Sven, please keep going, I didn't clean up all things in the wrapper (it not only impacts action_wrapper.vhd_source and also psl_accel_fgt.vhd_source ) 
And I am afraid I will not pick up HLS example on NVMe in the following one or two weeks.

boekholt, 2017-07-07:
Hi Lu, I am surprised that psl_accel_fgt.vhd_source is affected by that.
Are you working on a branch that I could have a look into?

luyong6, 2017-07-09:
Branch:  HLS_nvme_wrappers_update

Hi Sven,  I have cleaned up the irun compiling errors for a HLS generated NVME interface. 
Need to add parameters from top to bottom, so it impacts following files: 
```
hdl/core/psl_accel_types.vhd_source
hdl/core/psl_accel_fgt.vhd_source  
hdl/hls/action_wrapper.vhd_source
```
ARLOCK and AWLOCK are 2bits wire, and they are ingored in nvme_top. (Assigned "0") 

I created 
```
Hardware Part:  action_example/hls_nvme
Software Part:  examples/snap_hls_nvme.c
```
for the test. 
Currently just passing the irun compilation, and haven't touched the function.

luyong6, 2017-07-14:
Hi @boekholt How do you think this? Shall I raise a merge request?

boekholt, 2017-07-14:
@luyong6
Hi Lu,
is that already functional (do you have NVMe running with HLS)?
I thought this was not yet the case.
A prerequisite for merging would be to adapt the hdl_example action wrapper in order to prevent that it breaks.
Another point is that we are currently working on cloud provider support by adding a partial reconfiguration flow. This means that we have to get rid of the generics for the action wrappers and replace those with constants. This is almost done (there is already a merge request pending). I propose to merge the first one that is functional and adapt the other one before merging.

jsvogt, 2017-09-08:
action_wrapper.vhd_source is updated. Action hls_nvme_memcopy is available in master.
@joergkayser please verify

joergkayser, 2017-10-16:
hls_nvme_memcopy is part of my regression and works fine

closed 2017-10-16 by joergkayser