open-power/snap #341: more checks

ThomasFuchs, 2017-06-27:
To run SNAP gcc and xterm is  required, but the availability  will not checked.

fhaverkamp, 2017-06-28:
Can you make the title of this bug more descriptive, please?
Do you expect some dependency checking to be done? I mean libc, grep, perl, sed, ... maybe more. Do you really expect us to maintain such a list of dependencies some place? If this would be a distro package we could use the dependency checking for software some folks use autoconfigs "./configure" or alike. I could imagine that those tools/libs are located at different places for different distros making the checking complicated especially for libs.

fhaverkamp, 2017-06-28:
I do not see for which purpose we need xterm?

ThomasFuchs, 2017-06-30:
simulation requires a xterm

jsvogt, 2018-03-01:
Xterm is listed in the readme now. Is this good enough, so we can close this?

closed 2018-03-09 by ThomasFuchs