open-power/snap #347: hls_intersect:  snap_intersect -m1 error

joergkayser, 2017-06-29:
branch:     master d112fa2 2017-06-29

irun with KU3+BRAM works
    snap_intersect -m1 -v -t300
    snap_intersect -n1 -v -t600
    snap_intersect -n2 -v -t1200
    snap_intersect -I -m1 -v -t300
irun with KU3+DDR3 fails in snap_intersect -m1 -v -t300
    Start Step1 (Copy source data from Host to DDR) ..............
    Step 1 took 3851159 usec
    Start Step3 (Do intersection in DDR) ..............
    WARNING:get_bytes_silent:Socket disconnect on recv!
    WARNING:Socket failure getting PSL event!
    err: job execution -1: No such device!
xsim with KU3+DDR3 fails in snap_intersect -m1 -v -t300
    Start Step1 (Copy source data from Host to DDR) ..............
    Step 1 took 44121334 usec
    Start Step3 (Do intersection in DDR) ..............
    err: job execution -6: Timer expired!

luyong6, 2017-06-29:
Oh! I missed that! 
The problem is because, in HLS code I separated two methods. I used  `#define USE_HASH` or `#define USE_SORT` to distinguish two methods. They cannot be called at the same time. I did that when I was blocked by the timing issues. 

Let me try to open both of them again, otherwise I need add someting in the testing scripts.  

So, yes, current master code defines `USE_SORT`,  so `-m1` which means hash method will fail. This is as expected.

fhaverkamp, 2017-07-03:
Lu Yong, we briefly discussed this issue in our meeting here in BOE and it appears as if you build multiple different bitstreams. I wonder if this is a case where you should use a different action type, such that the query of the type will lead to the right test-cases and that the content is clear when reading out the type. So Eberhard has put some test cases in place which ask for the id and depending on it select the right set of tests. I think it would be good if that is possible for your code too. Please have a look, if my assumption is right. Thanks.

luyong6, 2017-07-03:
Hi Frank, I just started an image build so see if I can keep both method in one action. Tomorrow morning I can see the timing results.  I will check the results tomorrow morning. If it cannot be successful, I have to separate them to two IDs. Where are Eberhard's testcases?

joergkayser, 2017-07-04:
software/scripts/snap_jenkins.sh shows an example, where the action_type points to the right set of testcases:
MY_ACTION=`./software/tools/snap_maint -C $card -m 1`
for action in $MY_ACTION ; do
  case $action in
	*"10141005")
		test_hls_intersect $card $accel ;;
In your case you define multiple variables to distinguish the methode, but they cannot be separated as they all use the same Action_type=10141005

joergkayser, 2017-07-04:
hardware/action_examples/hls_intersect/action_intersect_hls.H always defines the sort method with USE_SORT, and the ifdef USE_HASH condition is buried in the else_path and never called.
Please describe how to select the hash method and how to determine, which method is built into the action

joergkayser, 2017-07-06:
@luyong6 have you decided on using different action IDs for your different build environments ?

luyong6, 2017-07-06:
Hi Joerg, my recent synthesis still failed with about -400ps, so I am going to seperate them to two IDs. I think I can keep the same software, when -m1, setting one ID,  and -m2 setting another ID. 

And under hardware/action_examples/ I am going to seperate the codes to *two folders* `hls_intersect_bysort` and `hls_intersect_byhash`. 
Currently hls_intersection uses 0x10140005, and I am also going to occupy 0x10140006 also. 
Your opinion? 

This week I am on travel so I will try to do this in weekend or early next week.

jsvogt, 2017-07-18:
>> Currently hls_intersection uses 0x10140005, and I am also going to occupy 0x10140006 also.
For new action types, please go ahead, change the table, and let someone from the team review the pull request.

joergkayser, 2017-07-18:
When possible, please also add the testcase and/or testcase options for the respective action

luyong6, 2017-07-20:
Hi I finished the testcase split in Branch HLS_intersect_split. 
I stuggled a long time with "make image" so it delayed so long. But how about I go to merge it after your directory restructure? (And also update Jenkin test) So I am afraid to keep  this issue opened for another 2-3 days.

joergkayser, 2017-07-20:
This branch HLS_split_intersect works fine, and you did a perfect job modifying our simulation regression 👍 
Go and push this to master right after the restructure merge

joergkayser, 2017-07-28:
with HLS_split_intersect_new being merged into master, this problem is solved. Thx

closed 2017-07-28 by joergkayser