open-power/snap #353: 64 byte output data swap

ReinerRieke, 2017-07-03:
Once in a while we see that the upper and lower 64 bytes in a cache line are swapped. I will take out an earlier 64 byte swap fix in order to see if that changes anything!

jsvogt, 2017-07-07:
Circumvention removed, testing

luyong6, 2017-07-10:
Be careful with the timing of ha_bwad..

ReinerRieke, 2017-07-18:
So far no 64 byte swap failures have been reported on KU3 and FGT. Will leave the issue open few more days.

ibm-genwqe, 2017-08-01:
See also https://github.com/open-power/snap/issues/387
We still found a Memory Swap or Data corruption problem on FGT Card. This problem is fixed
in Branch https://github.com/open-power/snap/tree/RC_flip_bit_4_fgt

jsvogt, 2017-08-01:
Well, not really fixed, but circumvented. Fix must go into the PSL

ThomasFuchs, 2017-10-06:
Fix went into PSL. Latest version is on IBM OpenPower Portal.

closed 2017-10-06 by ThomasFuchs