open-power/snap #356: Build: Elaboration error leads to multiple failing grep calls

fhaverkamp, 2017-07-04:
The problem below occurs if one uses an HLS action without CARD_DDR but configured
SDRAM_USED=TRUE. Elaboration fails and multiple calls to grep seem to go wrong.
I think we should check the return code and fail on the first one such that we stop on the
first occurring error.

````
****************************************************************************************
Now try BLOWFISH
****************************************************************************************
~/framework ~/framework
simulator=irun simdir=ies simtop=work.top
hostname=hdclv034 arguments=-explore -app examples/snap_blowfish -arg 
in sim_script subdirectory
prepare simout directory
cp: cannot stat `cds.lib': No such file or directory
cp: cannot stat `hdl.var': No such file or directory
SNAP_root=/afs/bb/u/haver/framework/snap
pslse_root=/afs/bb/u/haver/framework/pslse
simbase=   /afs/bb/u/haver/framework/snap/hardware/sim/ies
simout=    20170704_093253
run_sim sim_arg= -batch +model_data+. -64bit -input ncrun.tcl, tst_arg=  -v, AET=1
**** call simulator=irun in background, wait for socket in irun.log->shim_host.dat
irun(64): 15.10-s019: (c) Copyright 1995-2016 Cadence Design Systems, Inc.
irun: *E,RFAIL: A previous invocation failed while running the subexec ncelab.  -R/-r cannot be used until the design has been successfully compiled and elaborated.
Usage: grep [OPTION]... PATTERN [FILE]...
Try `grep --help' for more information.
Usage: grep [OPTION]... PATTERN [FILE]...
Try `grep --help' for more information.
Usage: grep [OPTION]... PATTERN [FILE]...
Try `grep --help' for more information.
Usage: grep [OPTION]... PATTERN [FILE]...
Try `grep --help' for more information.
Usage: grep [OPTION]... PATTERN [FILE]...
Try `grep --help' for more information.
Usage: grep [OPTION]... PATTERN [FILE]...
Try `grep --help' for more information.
Usage: grep [OPTION]... PATTERN [FILE]...
Try `grep --help' for more information.
Usage: grep [OPTION]... PATTERN [FILE]...
Try `grep --help' for more information.
Usage: grep [OPTION]... PATTERN [FILE]...
Try `grep --help' for more information.
Usage: grep [OPTION]... PATTERN [FILE]...
Try `grep --help' for more information.
tee sim.log PID=88294 ppid= sim_pid= after 10 sec
could not start irun
**** append msgs from sim.log
**** RC=-1 parallel=1 executed= failed= runtime= simtime= cpus=12 speed=2899 load=0.75 local=0 used_kB=42 percused=38%
**** Versions: SNAP=2ce94b1 2017-07-03 PSLSE=N/A
~/framework
bash-4.1$ 
 

9:57:53 AM: So sieht das aktuell aus. Die fehlschlagenenden greps sind doof, das sollten wir besser hinbekommen. 

9:57:57 AM: Das log sagt: 

9:58:14 AM: ncvlog: *W,SPDUSD: Include directory ../../viv_project/framework.ip_user_files/ipstatic/hdl given but not used.
        Total errors/warnings found outside modules and primitives:
                errors: 0, warnings: 1
        Elaborating the design hierarchy:
                Caching library 'work' ....... Done
                Caching library 'xil_defaultlib' ....... Done
                Caching library 'unisim' ....... Done
                Caching library 'unisims_ver' ....... Done
                Caching library 'secureip' ....... Done
                Caching library 'unimacro' ....... Done
                Caching library 'unimacro_ver' ....... Done
                Caching library 'blk_mem_gen_v8_3_5' ....... Done
                Caching library 'fifo_generator_v13_1_3' ....... Done
                Caching library 'axi_infrastructure_v1_1_0' ....... Done
                Caching library 'axi_clock_converter_v2_1_10' ....... Done
                Caching library 'xpm' ....... Done
ncelab: *E,CUCGNA: component generic HLS_ACTION.C_M_AXI_CARD_MEM0_ID_WIDTH not associated with any formal [5.2.1.2].
           Default binding for instance: top.a0@psl_accel(psl_accel):action_w@action_wrapper(STRUCTURE):hls_action_0
ncelab: *E,CUCPNA: component port HLS_ACTION.M_AXI_CARD_MEM0_AWADDR not associated with any formal [5.2.1.2].
           Default binding for instance: top.a0@psl_accel(psl_accel):action_w@action_wrapper(STRUCTURE):hls_action_0
ncelab: *W,CUDEFF: default binding failed for component instance (top.a0@psl_accel(psl_accel):action_w@action_wrapper(STRUCTURE):hls_action_0) with design unit (WORK.HLS_ACTION:BEHAV) because of the above mentioned reason.
ncelab: *W,CUNOTB: component instance is not fully bound (top.a0@psl_accel(psl_accel):action_w@action_wrapper(STRUCTURE):hls_action_0) [File:action_wrapper.vhd, Line:337].
irun: *E,ELBERR: Error during elaboration (status 1), exiting.
````

joergkayser, 2017-07-05:
@fhaverkamp  Frank, the data is inconsistent. You mentioned the elaboration error, but the grep and other messages come from simulation (run_sim call).
You should have received an error from make config (elaboration), which should be investigated, and not proceeding into simulation.

joergkayser, 2017-07-11:
please verify with branch=fix_grep_356

fhaverkamp, 2017-07-18:
I trust you. Thanks for taking care. If it reoccurs, I will open a new bug.

closed 2017-07-18 by fhaverkamp