open-power/snap #367: Build: make clean does not tidy up properly

fhaverkamp, 2017-07-06:
Hi Thomas,

bash-4.1$ git status | grep -v blowfish
# On branch HLS_move_4KiB_buffer_support
# Untracked files:
#   (use "git add <file>..." to include in what will be committed)
#
#	hardware/hdl/core/mmio.vhd
#	hardware/hdl/core/psl_accel.vhd_source
#	hardware/hdl/core/psl_accel_types.vhd
#	hardware/hdl/core/snap_core.vhd
#	hardware/hdl/core/snap_core_types.vhd
#	hardware/hdl/hls/action_wrapper.vhd

I think a make clean should leave the directory in a clean state. Please have a look if you can
include those files into the cleanup procedure or mark them in the .gitignore, which I think is 
here not appropriate.

Regards

Frank

jsvogt, 2017-07-07:
Please retry with clean repository.

fhaverkamp, 2017-07-21:
Sven reported that it works for him after our restructuring activity.

fhaverkamp, 2017-07-21:
https://github.com/open-power/snap/tree/SIM_Fix_Cleanup_if_SNAP_ROOT_is_not_set

When SNAP_ROOT is not set, the Makefile in the hardware subdirectory is doing funny things ... and not what it should do. I tried to attack that in the branch mentioned above.

fhaverkamp, 2017-07-28:
For me it works. Sven if you like to do further improvements, please go ahead.

boekholt, 2017-09-08:
Looks good since a while

closed 2017-09-08 by boekholt