open-power/snap #380: SNAP Action: Discuss function-id versus multi-action

fhaverkamp, 2017-07-12:
We observed that folks who like to do more complex things with an action tend to split the functionality into different pieces e.g. SET_KEY, CYPHER, DECYPHER. This makes sense and they currently do this by using a mode flag in the SNAP job description. Since this is unique per action, it is not easy to have examples which make sense on the first glance. Furthermore it leads to trying to reuse the snap job description for multiple purposes and that in turn becomes easily messy.

Multi action might be a way out of that situation, but maybe introducing a "function id" as part of the job interface can help to implement something without having multi action, which is quite heavy weight on the hardware side.

So action code could distinguish the job description by an function id ... but how to reflect that in the action HLS interface, Bruno?

Maybe the extension pointer helps to have a solution? So job description is just a pointer and we move the data into one or the other specific job description struct?

fhaverkamp, 2017-07-12:
I wonder if we run into the casing problem again, or if we can potentially have multiple hls_action_func0, 1, 2 ... with appropriate pragmas, which map different job-descriptions into our MMIO space.

bmesnet, 2017-07-21:
Trying to understand if this is to reduce the size, simplify the SNAP job description, or reduce the size code of the action. 
If it just to differentiate the code in HLS, I guess we can use the FUNCTION_INSTANTIATE pragma in HLS which helps differentiating code and is intended to help doing local optimization (not tested yet), but this requires that the parameter in the function call is fixed.
_void foo_sub(bool mode){
#pragma HLS FUNCTION_INSTANTIATE variable=mode
if (mode) {
// code segment 1
} else {
// code segment 2
} }
void foo(){
#pragma HLS FUNCTION_INSTANTIATE variable=select // I guess it should be mode rather than select
foo_sub(true);
foo_sub(false);
}_

fhaverkamp, 2018-07-11:
I think we should close this. And if the requirement pops up again, we can open a new issue to track the effort. Is there anything against closing this?

bmesnet, 2018-07-11:
yes agree for closing

closed 2018-07-11 by fhaverkamp