open-power/snap #395: SNAP not supporting IP from Xilinx IP catalog if an HLS action requires it

bmesnet, 2017-07-25:
_For this issue, an example of hls_code that is not yet integrated into SNAP can be found in HLS_bloomfilter branch > **action hls_bloom**_

With current SNAP release, the **create_run_hls_script.sh** generates vhd files in hls_action/hlsXXX/yyy/**syn**/vhdl directory. 
If IP from Xilinx IP catalog are needed, then  **tcl files** are also added in this same directory, but not taken in account yet by SNAP framework. Indeed, when using a math function (ceil, log,...) for example, HLS needs to access math IP from the Xilinx IP catalog.  
=> The 2 scripts **create_framework.tcl** and **create_ip.tcl** need to be modified to get those xci files added.
2 solutions to get those IP/xci files:
- create the xci files from these **tcl files** generated by HLS in **syn**/vhdl directory 
- uncomment the **export_design** line in **create_run_hls_script.sh** file to create a new **impl** directory (at same level than **syn**) and xci files can be imported from impl/ip/tmp.srcs/sources_1/ip/*/*.xci. _Be aware that this solution creates much more files (unneeded) than first solution_

Solution proposed by Lu Yong is the following:
in **create_ip.tcl** add just after the last ip creation:
  puts "                                generating HLS specific IPs"
  foreach x [glob -dir $act_root *.tcl] {
      source $x >> $log_file
      foreach y [glob -dir $usr_ip_dir *] {
          set z [glob -dir $y *.xci]
          puts $z
          export_ip_user_files -of_objects [get_files $z] -no_script -force >> $log_file
          export_simulation -of_objects [get_files $z] -directory $ip_dir/ip_user_files/sim_scripts -force >> $log_file
}
}
in **create_framework.tcl** add just after the importing IPs section:
puts "                          importing HLS action specific IPs"
foreach y [glob -dir $usr_ip_dir *] {
   set z [glob -dir $y *.xci]
   #puts $z
   add_files -norecurse $z >> $log_file
   export_ip_user_files -of_objects [get_files "$z"] -force >> $log_file
}
Discussion with Thomas that location to put these modifications need to be discussed with the SNAP dev team

luyong6, 2017-08-28:
Bruno, some updates to the scripts for your reference: 

create_ip.tcl
```
set act_root     $action_dir
append act_root  hw/$fpga_dir/$sol_name/syn/vhdl
set usr_ip_dir   $root_dir/ip/managed_ip_project/managed_ip_project.srcs/sources_1/ip

......

puts "	\[CREATE_VHDL_IPs..........\] start"
#put $act_root
foreach x [glob -nocomplain -dir $act_root *.tcl] {
    source $x >> $log_file
}
foreach y [glob -dir $usr_ip_dir *] {
        set z [glob -dir $y *.xci]
        #puts $z
        #generate_target {instantiation_template} [get_files $z] >> $log_file
        generate_target all                      [get_files $z] >> $log_file
        export_ip_user_files -of_objects [get_files $z] -no_script -force  >> $log_file
        export_simulation -of_objects [get_files $z] -directory $ip_dir/ip_user_files/sim_scripts -force >> $log_file
}
puts "	\[CREATE_VHDL_IPs..........\] done"
```

create_framework.tcl
```
set usr_ip_dir  $ip_dir/managed_ip_project/managed_ip_project.srcs/sources_1/ip

......

foreach y [glob -nocomplain -dir $usr_ip_dir *] {
   set z [glob -dir $y *.xci]
   #puts $z
   add_files -norecurse $z >> $log_file
   export_ip_user_files -of_objects  [get_files "$z"] -force >> $log_file
}

```

ThomasFuchs, 2017-09-11:
Hi Bruno, Lu Young,
the solution for this problem looks good. Please create a new branch/pull request to bring the code into github. The test and fine tuning could be done with the pull request.

boekholt, 2017-10-02:
I did implement and test the suggested solution. Please check pull request #472.

bmesnet, 2017-10-03:
Tested it ok with hls_bloomfilter action which is using math libraries and hls_helloworld which doesn't. Great work Sven and thanks to Lu Yong who gave the way to do it
Closing the issue

closed 2017-10-03 by bmesnet