open-power/snap #412: PR flow: Need to define Configuration Management for cloud workflow

boekholt, 2017-08-07:
Do we want to define a naming convention for the static region DCPs?

kenhill, 2017-10-23:
@boekholt, I currently include the static DCPs in a directory structure similar to:

`ls static_dcps`
ADKU3_SDRAM
N250S_SDRAM_NVME

boekholt, 2017-10-24:
@kenhill does that mean, that you need additional support here?
The idea behind the current implementation is to store all generated DCPs to the directory that `DCP_ROOT` is pointing to. Later the required DCPs (static and user) should get included from that directory.
The latest change that you suggested forces the user to manually set `DCP_ROOT` such that it points to a valid directory.
Do you require different directories for different DCPs?

kenhill, 2017-10-24:
@boekholt no, the current layout works. I'm assuming most cloud providers will provide the static DCPs that are supported on their platform. So the user will link/copy the static DCP image they want to use into `DCP_ROOT`

The build should fail in the Xilinx tools if incorrect DCPs are used. Should we close this issue?

boekholt, 2017-10-24:
@kenhill  If you think that the config management support that the SNAP framework does provide is sufficient for the cloud flow, then we may close this issue. (The static DCP naming topic was just an initial question for the general question regarding config mgmt support for the cloud flow.)

kenhill, 2017-10-24:
The defconfig files are enough to configure the current `PRFLOW`

closed 2017-10-24 by kenhill