open-power/snap #415: HLS_INTERSECT: Memory overflow during simulation

joergkayser, 2017-08-10:
When running snap_intersect  -m1 -v -t300, I get the error:
memory_write: at time 86449534.0 ps ERROR: Memory overflow.  Write to Address 0400000 with Data xxxxxxxx4141412b will be lost. You must increase the MEM_BITS parameter or define MAX_MEM.
...snap/hardware/ip/ddr3sdram_ex/imports/ddr3.v:756                 if (STOP_ON_ERROR) $stop(0);
ncsim>  exit

This means, that the Xilinx DDR3 IP needs additional information, how much memory is being addressed during simulation. Can you tell from the testcase and provide this to synthesis ?

luyong6, 2017-08-14:
I Joerg, in newer releases, I reduced the usage of DDR model so this issue may not show up. But yes, this is always a problem. My solution was to open the first line `define MAX_MEM in ddr3.v, but this is an extra step and not good for auto-testing.

joergkayser, 2017-09-12:
defining MEM_BITS to 12 (2^12 access to DDR memory) is better than defining MAX_MEM (writing to one distinct file, which prevents concurrent testing of multiple configs). See #415 If more is needed, please reopen

PR #437 (2017-09-12): Sim: fix for issue #415
fixed #415 

SIM: set MEM_BITS to 12 to allow more then 2^10 write data bursts
closed 2017-09-12 by joergkayser