open-power/snap #420: Running on FPGA: Action didn't exit clearly with middle termination

luyong6, 2017-08-16:
An example running on KU3 card (Also happened on NSA121B card) 
1. Running Sponge, SUCCESS, it costs 28seconds.
2. Set -t10, run it and let it be timed out after 10 seconds. 
3. Run again, (with this parameter it needs even shorter time, < 1second), but it timed out again. 
Then I cannot operate on this Acceleration anymore, always timeout. Unless reprograming the FPGA.

It seems the action doesn't exit clearly under unexpected situations (like App Ctrl-C, timout count hit)   

```
jab@antipode:~/luyong/test_intersect/snap/actions/hls_sponge/sw$ sudo ./snap_checksum -mSPONGE -I -t200 -cSPEED -n2 -f32 -C1
PARAMETERS:
  type_in:  0
  addr_in:  0000000000000000
  size:     00100000
  checksum_start: 0000000000000000
  mode:     00000002 SPONGE
  test_choice:00 SPEED
  nb_elmts:    00000002
  freq:    00000032
  job_size: 56 bytes
------------------
RETC=102 => SUCCESS
CHECKSUM=edac5ab5c53f3195
NB_TEST_RUNS=65536
NB_ROUND=65536
28457381 usec
------------------
268435456 Keccak-p[1600,24] calls
9432893.912 Keccak-p[1600,24] / Second.
```

```
jab@antipode:~/luyong/test_intersect/snap/actions/hls_sponge/sw$ sudo ./snap_checksum -mSPONGE -I -t10 -cSPEED -n2 -f32 -C1
PARAMETERS:
  type_in:  0
  addr_in:  0000000000000000
  size:     00100000
  checksum_start: 0000000000000000
  mode:     00000002 SPONGE
  test_choice:00 SPEED
  nb_elmts:    00000002
  freq:    00000032
  job_size: 56 bytes
err: job execution -6: Timer expired!
```

```
jab@antipode:~/luyong/test_intersect/snap/actions/hls_sponge/sw$ sudo ./snap_checksum -mSPONGE -I -t200 -cSPEED -n2 -f8192 -C1
PARAMETERS:
  type_in:  0
  addr_in:  0000000000000000
  size:     00100000
  checksum_start: 0000000000000000
  mode:     00000002 SPONGE
  test_choice:00 SPEED
  nb_elmts:    00000002
  freq:    00008192
  job_size: 56 bytes
err: job execution -6: Timer expired!
```

jsvogt, 2017-09-08:
Duplicate of #418 and #155

jsvogt, 2017-09-08:
Feature currently not in plan to be implemented

jsvogt, 2018-03-01:
@luyong6 Please check if this issue was fixed by pull request #574.

luyong6, 2018-03-02:
I rebuilt a hls_sponge and Today on 8a967b947f657e45a1f52f32e31a0e7799c7c1c5
and the above issue was solved perfectly. 
This issue can be closed.

closed 2018-03-02 by luyong6