open-power/snap #424: XSIM reports parity error while NCSIM doesn't

luyong6, 2017-08-29:
When using XSIM to simulate the basic hdl_example, user may see WARNING like this: 

```
Warning: FIR: HA_B write data parity error
Time: 300086 ns  Iteration: 2  Process: /top/a0/snap_core_i/dma/line__2011  File: /afs/vlsilab.boeblingen.ibm.com/u/luyong/capi/vol2/snap121/hardware/hdl/core/dma.vhd
```

But this error doesn't happen when setting SIMULATOR=irun. 

From the xsim waveform, I saw AC_GENPARITY may be modeled incorrectly in XSIM. It only lasts for one cycle. 
[image]

This may be annoying for user who only has XSIM as the simulator. @bmesnet , please be aware of this. And @ThomasFuchs do you think it is necessary to fix, or report bug to Xilinx?

luyong6, 2017-08-29:
ACTION_ROOT=...hdl_example, with SIMULATOR=XSIM
run_sim -list testlist.sh
(I think not related to FPGACARD and SDRAM_USED) 

The code is at line 498 of hdl/core/dma_buffer.vhd

```
495           -- write data checker
496           --
497           -- ha_b.wpar have a one cycle delay
498           rram_wdata_p_q <= AC_GENPARITY(ha_b_q.wdata, 64);
499
500           -- FIXME: please fix the bit swap in the AC_GENPARITY function
501           IF ha_b_wvalid_q  = '1' THEN
502              ha_b_wdata_err_q <= (rram_wdata_p_q(7) XOR ha_b_q.wpar(0)) &
503                                  (rram_wdata_p_q(6) XOR ha_b_q.wpar(1)) &
504                                  (rram_wdata_p_q(5) XOR ha_b_q.wpar(2)) &
505                                  (rram_wdata_p_q(4) XOR ha_b_q.wpar(3)) &
506                                  (rram_wdata_p_q(3) XOR ha_b_q.wpar(4)) &
507                                  (rram_wdata_p_q(2) XOR ha_b_q.wpar(5)) &
508                                  (rram_wdata_p_q(1) XOR ha_b_q.wpar(6)) &
509                                  (rram_wdata_p_q(0) XOR ha_b_q.wpar(7));
```

joergkayser, 2017-08-29:
The difference between xsim and irun is just, that you see the msg printed in xsim, whereas it is suppressed in irun (with assertion off) . Both simulators show the parity error in the waveform, and yes, it is annoying.
@ThomasFuchs Please fix the parity error, so that we can turn assertions back on on both simulators

ThomasFuchs, 2017-08-30:
The branch Issue#424 includes the fix.

PR #426 (2017-08-31): HW: fixed the ha_b_wdata parity checker
This change should fix #424
closed 2017-08-31 by ThomasFuchs