open-power/snap #432: SNAP enhancements

bmesnet, 2017-09-05:
Few ideas to make life easier:
1) Can we rename snap/actions to snap/examples since application and actions are now under this directory?
2) Can we have defined 0x1=cpu and 0x0=fpga so that SNAP_CONFIG=cpu will be more explicit than SNAP_CONFIG=0x1?
3) Can all tools checking or displaying **Action Types** (snap_maint, make snap_env) read the Type from the ActionTypes.md file rather than coding it in the tool so that adding a new type is immediately reflected by tools ?

ibm-genwqe, 2017-09-05:
Idea 3:
agree. I think we can create a C Header file like
struct ven_tab mytab[] = {
	0x10140000, "IBM Example",
	0x10140001, "IBM Example 1",
	0x10140002, "IBM Example 2",
	0x10140003, "IBM Example 3",
	0x10130Ab0, "xx Example 1",
	0x10150cd0, "xy Example 1",
	0, "exit",
};
this can be included in snap maint at compile time.

luyong6, 2017-09-06:
idea 2: 
agree. I also suggest to change the name "simulation_enabled()" in snap.c. This is not simulation, this is software action (SNAP_CONFIG=cpu) used. Add @fhaverkamp 

https://github.com/open-power/snap/blob/master/software/lib/snap.c

acastellane, 2017-09-07:
Hallo,

when running snap-env from scratch, if you make a mistake, you can't recover situation as it seems .snap-env.sh, once created is read but is not modified.

Process to duplicate issue :
new linux account on supervessel
git clone snap dir
setup variables :
    export xilinx_root=/opt/Xilinx                             <<=== this one should be in the readme.md file
    source $xilinx_root/Vivado/2016.4/settings64.sh
    export XILINXD_LICENSE_FILE=2100@cloudsynthesismaster
    export PSL_DCP=/mnt/cloud-data/CAPI_PSL_Checkpoints/KU3_Checkpoint/b_route_design.dcp
make snap-env
get a msg saying no pslse dir is set

As I forecast to use simulation, I want this to be properly set.

git clone pslse dir
export PSLSE_ROOT=/home/opuser/pslse
make set-env 
MSG : "Path to PSLSE not set properly" !
although "echo $PSLSE_ROOT" provides : "/home/opuser/pslse"

from here, you can't progress unless you erase .snap-env.sh

Suggestion : Either inform/suggest on what to do or reread all variable ?

Danke sehr.

Alexandre

bmesnet, 2017-10-03:
Pull request #471 did solve idea 2 but forgot to implement Lu Yong's change on idea 2 - will do that later

bmesnet, 2018-02-13:
Summarizing items discussed in this issue:
1.  Can we rename snap/actions to snap/examples since application and actions are now under this directory? **WON'T FIX**
2.  Can we have defined 0x1=cpu and 0x0=fpga so that SNAP_CONFIG=cpu will be more explicit than SNAP_CONFIG=0x1? **DONE**
3.  Can all tools checking or displaying Action Types (snap_maint, make snap_env) read the Type from the ActionTypes.md file rather than coding it in the tool so that adding a new type is immediately reflected by tools ? **DONE**
4.  I also suggest to change the name "simulation_enabled()" in snap.c. This is not simulation, this is software action (SNAP_CONFIG=cpu) used. **CORRECTED pull request#**
5.  snap_env configuration mistake **DONE**

bmesnet, 2018-02-14:
Code implemented. Closing issue

closed 2018-02-14 by bmesnet