open-power/snap #434: Wrong PSL Checkpoint Files for the CAPI SNAP Design Kit?

razvnane, 2017-09-05:
Dear,

I get a synthesis error (see the attached log) when I try to create the image for the hdl_example. For some reason I get a mismatch in several ports used in base_img module, so I am wondering if this is a problem with my version of the PSL checkpoint file? 

I downloaded the version from 04/14/2017 available on the IBM Portal for OpenPower. 

[synth_design.txt](https://github.com/open-power/snap/files/1277616/synth_design.txt)

Best regards,
Razvan

ThomasFuchs, 2017-09-06:
Hi Razvan,
SNAP currently supports two cards. An Alpha-Data ADM-PCIE-KU3 and a Nallatech 250S-2T. Both cards have the same FPGA (KU60) but needs different PSL checkpoints. In the synth_design.txt I saw that you have setup SNAP for the Nallatech card. Do you have downloaded the PSL for this card?

PSL checkpoint (.dcp) files for KU3        => Alpha-Data ADM-PCIE-KU3
PSL checkpoint (.dcp) files for FlashGT  => Nallatech 250S-2T

razvnane, 2017-09-06:
Hi Thomas,

Which line shows that I have assigned the Nallatech card? 

I downloaded actually both checkpoints, but I use only the KU3. And I see the PSL_DCP variable pointing to the correct directory in the output of the SNAP configuration.

ThomasFuchs, 2017-09-07:
Hi Razvan,
each PSL_DCP needs a other VHDL top structure. This is the reason why you find in hardware/hdl/core two top structures. The psl_fpga_ku3.vhd_source and the psl_fpga_fgt.vhd_source. During the configuration phase, we pick up one and rename it to psl_fpga.vhd. Which one we pick up is controlled by the variable FPGACARD. It looks to me that you have downloaded the PSL_DCP for the KU3, but set the FPGACARD to FGT (this is the default). The reason why I think so, is that I saw errors about missing led_* signals. Only the FGT card have led_* signals. 

Note: We have simplify the config step, and pushed this changes into gitHub in the last days. So it will be a good idea to check out the latest master and start from scratch.  

RELEASE NOTE:
We just changed the way of how the SNAP framework is getting prepared and configured for FPGA image and simulation model builds.
Instead of directly using environment variables the setup for the `make` process now is based on files `.snap_env.sh` and `.snap_config.sh`. If they do not already exist, these files get generated via the `Makefile` targets `snap_env` and `snap_config`. These targets make use of the Linux kconfig method.
(*Note:* It is recommended to call `make` from the snap root directory.)

The documentation in https://github.com/open-power/snap/tree/master/hardware/README.md got updated.

In order to provide some backward compatibility, the script https://github.com/open-power/snap/tree/master/hardware/snap_settings got adapted. So, the "old" way of setting up the environment and the configuration for builds should still work.

*Please let us know if questions arise or if you run into problems.*

razvnane, 2017-09-07:
Hi Thomas,

That was the problem. After changing the FPGACARD variable in the configuration files, everything went smooth. Thanks!

Best,
Razvan

ThomasFuchs, 2017-09-07:
Hi Razvan,
I will close this issue. If you have future question, please open a new one issue or contact me directly.

Best,
Thomas

closed 2017-09-07 by ThomasFuchs