open-power/snap #468: Kconfig introduces undocumented dependencies to kernel sources

ldurdel, 2017-09-27:
On Ubuntu 16.04 LTS the initial snap configuration with all dependencies as listed in the README in place fails with the following error message:
```
${SNAP_ROOT} $ make snap_config
menuconfig: Setting up SNAP configuration
SNAP Build Options
  Linux:    /usr/src/kernels/4.4.0-96-generic
  Build:    /home/lwenzel/mp2017/snap_new/scripts/build
  Kconfig:  /home/lwenzel/mp2017/snap_new/scripts/Kconfig
  config:   ../.snap_config
  autoconf: ../software/include/autoconf.h
ERROR: Not able to generate SNAP config files via kconfig, because directory /usr/src/kernels/4.4.0-96-generic does not exist.
Makefile:26: recipe for target 'menuconfig' failed
make[2]: *** [menuconfig] Error 1
Makefile:112: recipe for target 'menuconfig' failed
make[1]: *** [menuconfig] Error 1
Makefile:126: recipe for target 'snap_config' failed
make: *** [snap_config] Error 1
```

Simply installing the kernel sources does not solve the problem, as on Ubuntu they reside in `/usr/src/linux-source-vvv` and not the expected `/usr/src/kernels/vvv`. It works with a symlink, but this should be documented properly or the kernel source discovery mechanism made more flexible.

luyong6, 2017-09-29:
Maybe use `cat /etc/issue` and do decision? 

```
capi:snap$ cat /etc/issue
Ubuntu 15.04 \n \l

```

boekholt, 2017-10-02:
We should to look into using the standalone version, again.

boekholt, 2017-10-06:
The dependency to the kernel sources will be removed when change #476 is merged into master.
Instead a standalone version of kconfig will get installed automatically.

boekholt, 2017-10-10:
Change request #476 got just merged into master.
This means that we do no longer have a dependency to the kernel sources. The `make` process is now automatically cloning a standalone version of kconfig from https://github.com/guillon/kconfig.
A comment in `README.md` about this (hidden) dependency is outstanding. This comment will be part of change request #475.

boekholt, 2017-10-11:
Comments got added to change request #475

luyong6, 2017-10-16:
I am working on following Git version.  On Ubuntu 15.04. It still needs to install *ncurses-devel*
I executed one line of `sudo apt-get install ncurses-dev` based on the web search result from 
https://askubuntu.com/questions/270381/how-do-i-install-ncurses-header-files
and `make snap_config` again. Then the blue GUI works.

capi:snap$ git show
commit 82c087dd15ce012dbf2fb244c5772560660b44d9
```
capi:snap$ make snap_config
menuconfig: Setting up SNAP configuration
SNAP Build Options
  Kconfig:          /home/capi/project/snap/snap/scripts/Kconfig
  Build directory:  /home/capi/project/snap/snap/scripts/build
  config:           /home/capi/project/snap/snap/.snap_config
  autoconf:         /home/capi/project/snap/snap/software/include/autoconf.h
Cloning kconfig from github.com/guillon/kconfig into /home/capi/project/snap/snap/scripts/build
Cloning into '/home/capi/project/snap/snap/scripts/build'...
remote: Counting objects: 131, done.
remote: Total 131 (delta 0), reused 0 (delta 0), pack-reused 131
Receiving objects: 100% (131/131), 192.12 KiB | 65.00 KiB/s, done.
Resolving deltas: 100% (48/48), done.
Checking connectivity... done.
Executing:
  make -C /home/capi/project/snap/snap/scripts/build -f GNUmakefile TOPDIR=. SRCDIR=. KBUILD_KCONFIG=/home/capi/project/snap/snap/scripts/Kconfig menuconfig
 *** Unable to find the ncurses libraries or the
 *** required header files.
 *** 'make menuconfig' requires the ncurses libraries.
 *** 
 *** Install ncurses (ncurses-devel) and try again.
 *** 
Makefile:203: recipe for target 'dochecklxdialog' failed
make[3]: *** [dochecklxdialog] Error 1
```

boekholt, 2017-10-16:
yes, the dependency to the ncurses library is documented in [README.md](https://github.com/open-power/snap/blob/master/README.md#dependencies).
The question would be if we should add more support for using the command line interface instead of the menu driven version for kconfig (or if we should look into the possibility to add a kind of automatic detection in order to decide which kconfig version to use)

jsvogt, 2017-10-20:
@ldurdel I think this is fixed now, please close or comment

ldurdel, 2017-10-27:
I tried the new version and everything (including the ncurses ui after installing `libncurses-dev` on Ubuntu) worked fine.

closed 2017-10-27 by ldurdel