open-power/snap #502: add the $SNAP_ROOT/software/tools to PATH variable

bmesnet, 2017-11-13:
To ease customer's experience, would it be possible to add the $SNAP_ROOT/software/tools to PATH ?
This would help using the **snap_maint** and the **snap_find_card** from anywhere in simulation but also while executing the action program on Power8 server
Thanks

boekholt, 2017-11-14:
What do you think would be the appropriate place to do that?

bmesnet, 2017-11-14:
@boekholt. Could this be done when setting/exporting the SNAP_ROOT variable ?

boekholt, 2017-11-16:
@bmesnet Pull request #505 is completed. Please close this issue if the change meets your requirements.
Thanks
Sven

bmesnet, 2017-11-16:
This is more than perfect and will help a lot the user. Thanks a lot @boekholt . Quick and efficient as usual. Thx. Closing the issue

closed 2017-11-29 by bmesnet