open-power/snap #520: PSL9: testcases dont compile with PSLSE branch capi2

joergkayser, 2017-11-29:
see /afs/bb/proj/fpga/framework/CI_ws/SNAP_SIM_matrix/ACT/hdl_example/CRD/N250SP/SIM/irun
checkout from SNAP(RC_psl9_capi2.0) and PSLSE(capi2)

Warning from PSLSE:
 [CC]	 afu_driver.o
In file included from /afs/bb/proj/cte/tools/cds/Incisiv/15.20.s18/tools/include/svdpi.h:80,
                 from afu_driver.c:26:
/afs/bb/proj/cte/tools/cds/Incisiv/15.20.s18/tools/include/svdpi_compatibility.h:48:2:
warning: #warning "INCISIV is currently compliant with DPI header file svdpi.h as per IEEE 1800-2005.
Starting IUS16.10, the default header file will be as per IEEE 1800-2012.
DPI applications which need to continue complying with IEEE 1800-2005 will need to compile with
the macro DPI_COMPATIBILITY_VERSION_1800v2005 starting in 16.1.
The difference in the two header files is in the declaration of the struct t_vpi_vecval."

Error from snap_example:
	[CC]	snap_example
/afs/vlsilab.boeblingen.ibm.com/proj/fpga/framework/CI_ws/SNAP_SIM_matrix/ACT/hdl_example/CRD/N250SP/SIM/irun/
snap/software/lib/libsnap.so: undefined reference to `cxl_event_pending'
collect2: ld returned 1 exit status
make[3]: *** [snap_example] Error 1

fhaverkamp, 2017-12-04:
The error you mention here should have been fixed by the merge mentioned here:
    https://github.com/ibm-capi/pslse/commit/cf9db2dbfa0e463fe8bc8f33b3c568973c875d0e
Please have a look if you correctly updated pslse and let us know if the problem still persists.

joergkayser, 2017-12-07:
I used the fix above and also removed all testcases, which align on 64B or have an odd #of 64B buffersize. This is needed, as P9 PSLSE does not support read_lock/write_unlock for this cases.
I now see timeout problems, which are related to PSLSE issue 91 (not responding with REORDER_PERCENT>0). I tried even with fastest.parms and still fail. Investigaiton underway

joergkayser, 2017-12-12:
The update in pslse has fixed the compilation errors. I close this ticket and will work on other testcase issues seperately

closed 2017-12-12 by joergkayser