open-power/snap #525: README(s) need some clarifications

bmesnet, 2017-11-30:
In https://github.com/open-power/snap/blob/master/README.md#dependencies:
- typo : As of now, **two** FPGA cards can be used with SNAP: -> should be **three** 
- can we separate what is needed for x86 environment and Power8 environment:
   + is libcxl downloadable for x86 dev environment from the link (or is it provided with PSLSE)?
   + dcp files + kconfig + pslse are for x86 only
   + capi-utils is for P8 only
- is it possible to have a list of all variables need to be set on x86 and Power9 for each tool we can install (VPI_USER_H_DIR, LD_LIBRARY_PATH, DENALI, XILINX_VIVADO,......). This would be really a huge saving time for user!

In https://github.com/open-power/snap/blob/master/hardware/doc/Bitstream_flashing.md:
- reference to **export FACTORY_IMAGE=TRUE** while we now use the **make snap_config**!
- it is written "Therefore, when that test functionality or the software interface changes, the factory bitstream may have to be updated, too." In case of an update, can we give user the way to use capi-utils to write the factory zone rather than using the Xilinx probe? 

In file https://github.com/open-power/snap/tree/master/hardware:
- after **export XILINXD_LICENSE_FILE=<pointer to Xilinx license>**, should we add also the XILINX_VIVADO variable which seems required ?

jsvogt, 2017-12-01:
typo: np to change from "two" to "three" - Semptian was already added by @luyong6  
There is no CAPI on x86, so the only libcxl there is for simulation, included in PSLSE  
Why should a user ever set VPI_USER_H dir, or LD_LIBRARY_PATH manually? The time saving is already done, because the only environment settings that must be set by the user are in snap_env.sh: PSL_DCP and PSLSE_ROOT. For existing actions, `make snap_config` sets the ACTION_ROOT in that script, for user-written actions it must be changed to point to the action directory.  

Xilinx_Vivado is set by the setup script required by Xilinx - as described in the hardware README:
`source <xilinx_root>/Vivado/<version>/settings64.sh`
We don't duplicate or mess with Xilinx environment settings, as they could change.

The factory image build instructions are going to change next week. `make snap_config` does work, but the GUI can't be scripted yet (defconfig needed). 
Users never need to update their factory image as long as there's a valid CAPI image. I will remove the comment about software interface changes - it only applies when a card is released with a factory load with guaranteed functionality for warranty reasons.  
The tools on https://github.com/ibm-capi/capi-utils don't support factory image programming on purpose (to change this, raise a bug there).

jsvogt, 2018-02-08:
I just merged the updated image build instructions into master. 
More updates to the Bitstream_flashing.md are in branch DOC_flashing.
https://github.com/open-power/snap/compare/DOC_flashing

bmesnet, 2018-02-13:
closing issue since correction done

closed 2018-02-13 by bmesnet