open-power/snap #526: MCS Hardware Setup Hard Coded Partitions

kenhill, 2017-12-01:
https://github.com/open-power/snap/blob/master/hardware/setup/build_mcs.tcl#L23

The build_mcs.tcl hard codes the values for `user` and `factory` partitions. This does not look correct for boards other than ADKU3.

N250S
https://github.com/ibm-capi/capi-utils/blob/master/src/capi_flash_nallatech_user.c#L145

Semptian
https://github.com/ibm-capi/capi-utils/blob/master/src/capi_flash_semptian_user.c#L150

jsvogt, 2017-12-04:
N250S is NOT using the capi_flash_nallatech_user.c, that's for the Altera version.
See the Makefile:
`capi-flash-NallatechKU60: src/capi_flash_ad7v3ku3_user.c`
That is the same as for ADKU3, and has been working nicely.
=> reject for N250S.

Rerouting to @luyong6 for Semptian

luyong6, 2017-12-05:
Hi Ken, 
Semptian told me if using their own factory image (not CAPI one), they use Xilinx ICAP function to jump from address 0 to address 400_0000(byte) to load user image. 

I have programmed two different *CAPI* image to address 0(image A) and address 400_0000(image B) but the working one is always image A. That means the factory image doesn't make a jump to the user image. 

I looked at the code in PSL, and haven't figured out the mechanism of how a CAPI image jump from factory to user. 

Currently in capi-utils, it writes address 0 (factory space), it doesn't have issues to work functionally. 

But you are right, build_mcs.tcl is not correct for S121B. Line 23 should be "up 0x2000000" for S121B because its user space is at 64MByte position. But even after I modified this, This card still has problem to jump from factory to user image now.

jsvogt, 2017-12-05:
Xilinx Multi-Boot documentation (for BPI, there's another one for SPI) https://www.xilinx.com/support/documentation/application_notes/xapp1246-multiboot-bpi.pdf
ICAP is embedded in PSL and can be used to switch to the factory side. For this purpose, the PSL  sets the address to load from in ICAP and triggers a reload (search for wbstart_addr in PSL source). The addresses given to ICAP by PSL must match the Flash layout of the card.
Note there was a defective PSL (fixed in May) which didn't reload the card when requested through sysfs, but it just reset the logic and left the old image in the FPGA => please check.

jsvogt, 2018-03-21:
Need more information, such as a users guide, from the manufacturer which specifies how the multiboot scenario must be implemented.

Which Flash page does the card boot from by default (address 0, factory image?). Then how does it expect the bitstream image to switch to the user image?
What are the addresses, and which multiboot method is required?
Is that the scenario described in figure 2 of https://www.xilinx.com/support/documentation/application_notes/xapp1246-multiboot-bpi.pdf ?
Implementing that requires different parameters in snap_bitstream_pre.tcl for the NSA121B.

The other supported cards load the user image from the high address directly. It only falls back to the factory partition at address 0 in case of a configuration error (using the RS1:0 signals), or triggered by CAPI PSL through ICAP.

jsvogt, 2018-04-18:
@luyong6 Did you get the information from the card manufacturer by chance?

luyong6, 2018-04-19:
Hi the partition is similar to 8k5, 64MB

Because they are Ku115 and the image size may be doubted (compared to Ku060)

I have something not clear, the logic in PSL. Let me write in a separate email.

已从 IBM Verse 发送

   Joerg-Stephan Vogt --- Re: [open-power/snap] MCS build: add Semptian NSA121B partition addresses (#526) ---

jsvogt, 2018-07-11:
Meanwhile @luyong6 added the necessary updates to hardware/setup/S121B/snap_bitstream_pre.tcl and hardware/setup/flash_mcs.tcl for both NSA121B variants.

closed 2018-07-11 by jsvogt