open-power/snap #539: path documentation

acastellane, 2017-12-07:
Hello,
Could we have a note on sourcing snap_path.sh ? as it is the minimum required to compile. (particularly in the case of deployment on POWER machines)
Could we also mention that these paths are automatically set when running any make XXX ?
thanks

fhaverkamp, 2018-01-26:
For simulation we set the path's such that the binaries and libraries can be found automatically. In the application binaries we are setting the rpath such that the shared libs can be found automatically too. So I wonder what is left over, since I believe you documented the use of snap_path inclusion in the Nimbix documentation to make calling the applications and finding the libraries more convenient for the user.

acastellane, 2018-02-13:
I agree it is not convenient to make a note. It is useful to just source snap_path when we want to run a script like snap_find_card, which does not require any compilation. So user's guide will take this into account.
Thanks

closed 2018-02-13 by acastellane