open-power/snap #543: simulation of hdl_nvme_example: timeout writing into flash

joergkayser, 2017-12-11:
With action=hdl_nvme_example, I am trying to write 4kB into flash with snap_cblk -n1 -t1 -b1 --write asc.in -v and get timeout after 3x60sec. CBLK_REQTIMEOUT is set to 60sec.

sim output is:
NVMe device has 209715200 blocks of each 4096 bytes; 819200 MiB (MAX 0xc800000 blocks) @ 1 threads
Writing NVMe from asc.in
[check_req_timeouts] err: req[ 0]: WRITING 60/61 sec LBA=0 TIMEOUT
[check_req_timeouts] err: req[ 0]: WRITING 60/61 sec LBA=0 TIMEOUT
[check_req_timeouts] err: req[ 0]: WRITING 60/62 sec LBA=0 TIMEOUT

The sim.log counts tags from 0 to 1f and gets the responses from PSLSE 
I tried with fastest.parms (strict order, no PAGED, no extra R/W activity) and it still fails.
Has this ever been simulated ? If yes, what timeout do you expect ?
Do you need a data file with 4096B including or excluding the EOF ?

Results and AET are in /afs/bb/proj/fpga/framework/jkayser/snap_testlist_updates/hardware/sim/ies/20171211_174448

joergkayser, 2017-12-12:
When running with SNAP_TRACE=0xF2, I get following msgs from snap_cblk:
R   hw_snap_mmio_read32(0x19b8010, f04c, 0) 0
R   hw_snap_mmio_read32(0x19b8010, f04c, 0) 0
R   hw_snap_mmio_read32(0x19b8010, f04c, 0) 0
R   hw_snap_mmio_read32(0x19b8010, f04c, 0) 0
R   hw_snap_mmio_read32(0x19b8010, f04c, 10) 0
B 0000469a.000046b3 1513086682627618     [completion_status] HW WRITE_COMPLETED 00000010 slot: 0 LBA=0
B 0000469a.000046b3 1513086682627630   [completion_thread] waking up slot 0 LBA=0
B 0000469a.000046b3 1513086682628284 [completion_thread_cleanup] tid=18099 tidy up ...
B 0000469a.0000469a 1513086682628352 [cblk_close] id=0 req_status=IDLE work_in_flight=0 now: 1513086682 sec 628346 usec ...
R   hw_snap_mmio_write64(0x19b8010, 1010, 2)
Formatting of 4096 bytes took 25775846 usec @ 0.000 MiB/sec

Here the simulation hangs, until I cancel snap_cblk.
Timeout values are not recognized.
Can you please investigate your MUTEX ?

joergkayser, 2017-12-13:
please start investigating @ snap_cblk.c line 798, which produces the last trace msg:
printf(stdout, "Formatting of %lld bytes took %lld usec @ %.3f MiB/sec\n",
			(long long)num_lba * lba_size, (long long)diff_usec, mib_sec);
This is followed by rqueue_init, which contains pthread_mutex_init....... any sign of deadlock ?

fhaverkamp, 2017-12-15:
Formating should happen in --rw or in --format. Not in --write. When using --write you need to offer a file of the appropriate size. A block is 4KiB. So it needs exactly to be 4096 bytes (ls -l data.bin).
The code you mention is the --rw test. The mutex should not be problematic, it is used to determine the next block to read or write and each block is just read/written one time here. You have -n1 such that there should be only 1 block, with -t1 only one thread. Did you try something like -n16 yet?

joergkayser, 2017-12-15:
The test pgm initializes with snap_nvme_init, and this runs fine
The error is in snap_cblk --format or snap_cblk --write, unrelated to the content of the input file
options="-n1 -t1"
for blk in 1 2;do p8=$((blk*8)); p4k=$((blk*4096));
        echo "generate data for $blk blocks, $p8 pages, $p4k bytes"
        step "snap_nvme_init -d0 -v"
        step "snap_cblk $options -b${blk} --format --pattern INC -v"
#      step "snap_cblk $options -b${blk} --write asc.in         -v"
      done
The files are generated with
   dd if=/dev/urandom of=rnd.in count=${p8} bs=512 2>/dev/null  # random data
   head -c $p4k </dev/zero|tr '\0' 'x' >asc.in                                      # same char mult times

fhaverkamp, 2017-12-18:
The blocksize for the snap_cblk application is not 512 bytes, but 4KiB. So we need bs=4096. Or is your  printout above not showing the * correctly?

fhaverkamp, 2017-12-18:
Just talked to Sven, he said a brock write needs more than 3 min, so the timeout needs for sim e.g. to be 5min. He reviewed our code and found that this might be reduced a little bit by shifting the position in the code where we take the time for the timeout measurement. But I claim that this will not do more than increasing the overall timeout.

Here is what he suggested:

```
static void req_start(struct cblk_req *req, struct cblk_dev *c)
{
	uint8_t action_code = req->action & 0x00ff;
	int slot = req->slot;

	req->tries++;
	block_trace("    [%s] HW %s memcpy_%x(slot=%u dest=0x%llx, "
		"src=0x%llx n=%lld bytes) LBA=%ld attempts=%d\n",
		__func__, action_name[action_code % ACTION_CONFIG_MAX],
		req->action, slot, (long long)req->dst, (long long)req->src,
		(long long)req->size, req->lba, req->tries);

       /* Sven suggested to move the gettimeofday() calls to this position, since the simulator polls of the status register take a lot of time during the previous mmios which eats already quite an amount of timeout ticking */

	pthread_mutex_lock(&c->dev_lock);
	gettimeofday(&req->stime, NULL);
	gettimeofday(&req->h_stime, NULL);

	__cblk_write(c, ACTION_CONFIG,    req->action);
	__cblk_write(c, ACTION_DEST_LOW,  (uint32_t)(req->dst & 0xffffffff));
	__cblk_write(c, ACTION_DEST_HIGH, (uint32_t)(req->dst >> 32));
	__cblk_write(c, ACTION_SRC_LOW,   (uint32_t)(req->src & 0xffffffff));
	__cblk_write(c, ACTION_SRC_HIGH,  (uint32_t)(req->src >> 32));
	__cblk_write(c, ACTION_CNT,       req->size);

	/* Wait for Action to go back to Idle */
	snap_action_start(c->act);

	/*
	 * Update statistics under device lock.
	 * Otherwise it might be off a little bit.
	 */
	req->tries++;
	if (action_code == ACTION_CONFIG_COPY_HN) {
		c->hw_block_writes++;
		c->wbytes_total += req->size;
	} else {
		c->hw_block_reads++;
		c->rbytes_total += req->size;
	}

	pthread_mutex_unlock(&c->dev_lock);
}
```

joergkayser, 2017-12-18:
I increased the timeout to 6min and started snap_cblk. It got stuck at the same place and hung for one hour, before I killed the app. Please advise

fhaverkamp, 2017-12-18:
I checked if the real-world application works with the parameters you have set, to exclude of-by-one errors in the software.

```
haver@tul3eth3:~/snap/actions/hdl_nvme_example/sw$ time ./snap_cblk -C0 --format -b1 -n 0x1
NVMe device has 209715200 blocks of each 4096 bytes; 819200 MiB (MAX 0xc800000 blocks) @ 1 threads
Formatting NVMe drive 4 KiB with pattern ff ...
Formatting of 4096 bytes took 139 usec @ 29.468 MiB/sec

real	0m0.003s
user	0m0.000s
sys	0m0.000s

haver@tul3eth3:~/snap/actions/hdl_nvme_example/sw$ ./snap_cblk -n1 -t1 -b1 --format
NVMe device has 209715200 blocks of each 4096 bytes; 819200 MiB (MAX 0xc800000 blocks) @ 1 threads
Formatting NVMe drive 4 KiB with pattern ff ...
Formatting of 4096 bytes took 156 usec @ 26.256 MiB/sec
```

Mhm, well. It works there nicely.

fhaverkamp, 2017-12-18:
Let us enable all the debugging with SNAP_TRACE=0xfff to see how far we come here. I am curious how much of the statistic printouts we get since those are done in the library destructor.

joergkayser, 2017-12-19:
Tested again with branch=RC_hdl_nvme_fix and got the hang situation with SNAP_TRACE=0xFF
see /afs/bb/proj/fpga/framework/jkayser/snap_nvme/hardware/sim/ies/20171219_135525

fhaverkamp, 2017-12-19:
http://bencane.com/2011/09/22/kill-creating-a-core-dump/

Can we automatically create a core dump by setting ulimits -c unlimited and sending a kill -3 $pid?
And at the same time set SNAP_TRACE=0xfff to get some more debug printout?

fhaverkamp, 2017-12-20:
https://github.com/open-power/snap/tree/SIM_improve_thread_safety should fix the issue.

joergkayser, 2017-12-21:
fixed and merged into master

closed 2017-12-21 by joergkayser