open-power/snap #563: KConfig fails with new gcc version

luyong6, 2018-01-02:
Currently the gcc on bb server is version 4.4.7 20120313 (Red Hat 4.4.7-3)

But If we run with gcc (4.8.2) or above, kconfig would fail with the following error message: |  
-- | --
make -C /afs/awd.austin.ibm.com/usr/u9/shgoupf/capi/snap/scripts/build -f GNUmakefile TOPDIR=. SRCDIR=. KBUILD_KCONFIG=/afs/awd.austin.ibm.com/usr/u9/shgoupf/capi/snap/scripts/Kconfig menuconfig/opt/rh/devtoolset-2/root/usr/libexec/gcc/x86_64-redhat-linux/4.8.2/ld: ././lxdialog/checklist.o: undefined reference to symbol 'acs_map'/opt/rh/devtoolset-2/root/usr/libexec/gcc/x86_64-redhat-linux/4.8.2/ld: note: 'acs_map' is defined in DSO /lib64/libtinfo.so.5 so try adding it to the linker command line/lib64/libtinfo.so.5: could not read symbols: Invalid operationcollect2: error: ld returned 1 exit statusmake[3]: *** [mconf] Error 1 | 

The "new" gcc is here: 
/opt/rh/devtoolset-2/root/usr/bin/gcc 
4.8.2 20140120 (Red Hat 4.8.2-15)

fhaverkamp, 2018-01-16:
I installed a recent CentOS version in a docker container and tried to do a make snap_config ; make menuconfig. It complained a little bit about missing packages initially, but after I had installed them e.g. kernel-build and ncurses-devel, it was able to build the menuconfig system.

In your case it appears as if the linker (collect2) is unable to link all code properly together. Can you see which libtinfo you have installed, please (for whatever libtinfo is used finally):
```
[root@4368c4487939 snap]# ls -l /lib64/libtinfo.so.*
lrwxrwxrwx 1 root root     15 Jan  7 16:58 /lib64/libtinfo.so.5 -> libtinfo.so.5.9
-rwxr-xr-x 1 root root 174576 Sep  6 22:08 /lib64/libtinfo.so.5.9
```

fhaverkamp, 2018-01-16:
```
[root@4368c4487939 snap]# gcc --version
gcc (GCC) 4.8.5 20150623 (Red Hat 4.8.5-16)
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

fhaverkamp, 2018-01-16:
```
[root@4368c4487939 snap]# make config
config: Setting up SNAP configuration
SNAP Build Options
  Kconfig:          /root/snap/scripts/Kconfig
  Build directory:  /root/snap/scripts/build
  config:           /root/snap/.snap_config
  autoconf:         /root/snap/software/include/autoconf.h
Executing:
  make -C /root/snap/scripts/build -f GNUmakefile TOPDIR=. SRCDIR=. KBUILD_KCONFIG=/root/snap/scripts/Kconfig config
*
* Kernel Configuration
*
Card Type
> 1. Nallatech 250S with 4GB DDR4 SDRAM, NVMe and Xilinx KU60 FPGA (N250S)
  2. Nallatech 250S+ with 4GB DDR4 SDRAM, NVMe and Xilinx KU15P FPGA (N250SP)
  3. AlphaData KU3 Card with Ethernet, 8GB DDR3 SDRAM and Xilinx KU60 FPGA (ADKU3)
  4. Semptian NSA121B Card with Ethernet, 8GB DDR4 SDRAM and Xilinx KU115 FPGA (S121B)
```

Looks working for me.

shgoupf, 2018-01-22:
Hey Frank,

Thanks for your information.

However, I still have the following error in my env, as below:

ERROR:
```
/opt/rh/devtoolset-2/root/usr/libexec/gcc/x86_64-redhat-linux/4.8.2/ld: ././lxdialog/checklist.o: undefined reference to symbol 'acs_map'
/opt/rh/devtoolset-2/root/usr/libexec/gcc/x86_64-redhat-linux/4.8.2/ld: note: 'acs_map' is defined in DSO /lib64/libtinfo.so.5 so try adding it to the linker command li
ne
/lib64/libtinfo.so.5: could not read symbols: Invalid operation
```

libtinfo:
```shgoupf@lx7040.austin.ibm.com:$ ls /lib64/libtinfo.so.5 -l
lrwxrwxrwx. 1 root root 15 Nov 11  2015 /lib64/libtinfo.so.5 -> libtinfo.so.5.7*
```

gcc version:
```
shgoupf@lx7040.austin.ibm.com:$ which gcc
/opt/rh/devtoolset-2/root/usr/bin/gcc
shgoupf@lx7040.austin.ibm.com:$ gcc -v
gcc version 4.8.2 20131111 (Red Hat 4.8.2-4) (GCC) 
```

fhaverkamp, 2018-01-23:
Can you describe your environment a little bit? Why is the compiler installed in /opt..../root/usr/bin is that a normal installation or something special? Can I try this out myself, such that we can look at the issue together?

shgoupf, 2018-01-23:
Hey, this is from Austin AFS (EOD/LSF) machine, for which I believe you should have access. The reason for using a gcc in /opt/... is because the version in gcc distro is pretty old. This is a standard usage from AFS based LSF machines.

fhaverkamp, 2018-01-25:
Did you try to use a consistent updated Linux distribution? I mean supporting mixed stuff like old distribution with new extra installed compiler seems a little overkill to me and endless matter of issues e.g. libraries not really matching for the task you like to do. What do you think?

@shgoupf Nice update to your identity page with cool picture. Thanks.

shgoupf, 2018-01-25:
@fhaverkamp , Hey Frank. Thanks for your time on this. I have to admit that the problem is not reproducible on other stable linux distros (like ubuntu) with new gcc versions (such as 4.9.2). Seems like it is  a problem specific to the distro combination and gcc version on EOD/LSF. 

I am OK to leave it as it is and let's focus on other more important issues.

Thanks again.

jsvogt, 2018-01-25:
So, can this issue be closed as invalid?

luyong6, 2018-01-26:
Yes, thanks all.

closed 2018-01-26 by luyong6