open-power/snap #567: PSLSE latest release prevents SNAP from working ok

bmesnet, 2018-01-10:
PSLSE latest release ( c65870f) prevents SNAP from running since new variables (MAJOR_VERSION  / MINOR_VERSION) have been created and are not supported by SNAP yet.
**CIRCUMVENTION :** Using tag v3.1 from PSLSE solves the issue (git checkout v3.1)
**SOLUTION :** How can we work with PSLSE to never impact SNAP ? working with known PSLSE levels documented in SNAP Readme.md pslse near pslse link ?

joergkayser, 2018-01-10:
Bruno, this is a known issue with the current master branch from PSLSE, and this is documented in our README:

PSLSE setup
The SNAP framework's simulation depends on the PSL simulation environment (PSLSE).
You may clone PSLSE from github https://github.com/ibm-capi/pslse.
⚠️ Currently (as of Dec 2017) the following PSLSE GIT branches are necessary:
    v3.1 (an older but stable master version) for POWER8 cards (ADKU3, N250S, S121B)
    capi2 (a seperate side branch with more features) for POWER9 cards (N250S+)
The plan is to merge both functionalities to master in 1Q2018

And it is documented in the PSLSE GIT under https://github.com/ibm-capi/pslse/issues/99.
Sorry for the churn, which was created merging PSLSE P8 and P9 functionality

bmesnet, 2018-01-10:
Thanks Jörg
Can you point me to the README.md you are referring to ?  i looked to both of README.md and didn't find your reference to v3.1.
https://github.com/open-power/snap/blob/master/README.md
https://github.com/open-power/snap/blob/master/hardware/README.md
Can we add a comment in master/README.md where the link to pslse is displayed so that no ambiguity remains ? Thx

joergkayser, 2018-01-11:
The master/README.md shows following and points to hardware/sim/README.md:
For simulation, SNAP relies on the xterm program and on the PSL Simulation Environment (PSLSE) which is available on github: https://github.com/ibm-capi/pslse
Simulating the NVMe host controller including flash storage devices requires licenses for the Cadence Incisive Simulator (IES) and DENALI Verification IP (PCIe and NVMe). Building images is possible without this. For more info see the Simulation README.

The hardware/README.md has no information about PSLSE, but also points for all simulation related items to hardware/sim/README.md

We took care, that PSLSE is only described in one of the three README pages. If you think we should consolidate three to two or even one page, let me know, and I will discuss this with the team

bmesnet, 2018-01-12:
Thanks Jörg. This is exactly what I missed.
I would so just add in [snap/README](https://github.com/open-power/snap/blob/master/README.md) 
_For simulation, SNAP relies on the xterm program and on the PSL Simulation Environment (PSLSE) which is available on github **(For more info see the [PSLSE setup](https://github.com/open-power/snap/blob/master/hardware/sim/README.md#pslse-setup))**:_

    https://github.com/ibm-capi/pslse

jsvogt, 2018-01-12:
The top README already contains a note which links to the sim README:  
For more info see the [Simulation README](https://github.com/open-power/snap/blob/master/hardware/sim/README.md)  
The sim readme then contains the hint with a :warning: sign.
Isn't that sufficient?

bmesnet, 2018-02-13:
Changes done. Closing the issue

closed 2018-02-13 by bmesnet