open-power/snap #578: Vivado 2017.4: compile errors in hls_intersect

joergkayser, 2018-01-12:
when calling 'make' in actions/hls_intersect/hw, I get following compile errors with Vivado 2017.4:

****** Vivado(TM) HLS - High-Level Synthesis from C, C++ and SystemC v2017.4 (64-bit)
  **** SW Build 2086221 on Fri Dec 15 20:54:30 MST 2017
  **** IP Build 2085800 on Fri Dec 15 22:25:07 MST 2017
    ** Copyright 1986-2017 Xilinx, Inc. All Rights Reserved.

source /afs/bb/proj/fpga/xilinx/Vivado/2017.4/scripts/vivado_hls/hls.tcl -notrace
INFO: [HLS 200-10] Running '/afs/bb/proj/fpga/xilinx/Vivado/2017.4/bin/unwrapped/lnx64.o/vivado_hls'
INFO: [HLS 200-10] For user 'jkayser' on host 'hdcl070.boeblingen.de.ibm.com' (Linux_x86_64 version 2.6.32-696.10.2.el6.x86_64) on Fri Jan 12 16:50:52 CET 2018
INFO: [HLS 200-10] On os "Red Hat Enterprise Linux Workstation release 6.4 (Santiago)"
INFO: [HLS 200-10] In directory '/afs/vlsilab.boeblingen.ibm.com/proj/fpga/framework/jkayser/snap_master/actions/hls_intersect/hw_h'
INFO: [HLS 200-10] Opening project '/afs/vlsilab.boeblingen.ibm.com/proj/fpga/framework/jkayser/snap_master/actions/hls_intersect/hw_h/hlsIntersect_h_xcku060-ffva1156-2-e'.
INFO: [HLS 200-10] Adding design file 'hls_intersect_h.cpp' to the project
INFO: [HLS 200-10] Adding test bench file 'hls_intersect_h.cpp' to the project
INFO: [HLS 200-10] Opening solution '/afs/vlsilab.boeblingen.ibm.com/proj/fpga/framework/jkayser/snap_master/actions/hls_intersect/hw_h/hlsIntersect_h_xcku060-ffva1156-2-e/intersect_h'.
INFO: [SYN 201-201] Setting up clock 'default' with a period of 4ns.
INFO: [HLS 200-10] Setting target device to 'xcku060-ffva1156-2-e'
INFO: [HLS 200-10] Analyzing design file 'hls_intersect_h.cpp' ...
INFO: [HLS 200-10] Validating synthesis directives ...
INFO: [HLS 200-111] Finished Checking Pragmas Time (s): cpu = 00:00:30 ; elapsed = 00:00:16 . Memory (MB): peak = 438.656 ; gain = 13.375 ; free physical = 6806 ; free virtual = 46458
INFO: [HLS 200-111] Finished Linking Time (s): cpu = 00:00:32 ; elapsed = 00:00:18 . Memory (MB): peak = 438.656 ; gain = 13.375 ; free physical = 6807 ; free virtual = 46459
INFO: [HLS 200-10] Starting code transformations ...
INFO: [XFORM 203-603] Inlining function 'write_single' into 'check_table2' (hls_intersect_h.cpp:352).
INFO: [XFORM 203-603] Inlining function 'write_single' into 'make_hashtable' (hls_intersect_h.cpp:290).
INFO: [XFORM 203-603] Inlining function 'write_single' into 'make_hashtable' (hls_intersect_h.cpp:275).
INFO: [XFORM 203-603] Inlining function 'read_single' into 'check_table2' (hls_intersect_h.cpp:347).
INFO: [XFORM 203-603] Inlining function 'read_single' into 'check_table2' (hls_intersect_h.cpp:339).
INFO: [XFORM 203-603] Inlining function 'read_single' into 'make_hashtable' (hls_intersect_h.cpp:280).
INFO: [HLS 200-111] Finished Standard Transforms Time (s): cpu = 00:00:34 ; elapsed = 00:00:20 . Memory (MB): peak = 567.824 ; gain = 142.543 ; free physical = 6776 ; free virtual = 46428
INFO: [HLS 200-10] Checking synthesizability ...
INFO: [XFORM 203-602] Inlining function 'ht_hash' into 'make_hashtable' (hls_intersect_h.cpp:265) automatically.
INFO: [XFORM 203-602] Inlining function 'ht_hash' into 'check_table2' (hls_intersect_h.cpp:334) automatically.
INFO: [XFORM 203-602] Inlining function 'compare_eq' into 'check_table2' (hls_intersect_h.cpp:349) automatically.
ERROR: [SYNCHK 200-11] Global Variable 'entry_t' has an unsynthesizable struct type '%struct.entry_t.1.107.134 = type { [64 x i8], %struct.entry_...' (a member pointer to struct itself).
WARNING: [SYNCHK 200-23] hls_intersect_h.cpp:218: variable-indexed range selection may cause suboptimal QoR.
INFO: [SYNCHK 200-10] 1 error(s), 1 warning(s).
ERROR: [HLS 200-70] Synthesizability check failed.

The same compile works fine for Vivado 2017.2
Can you please check your sources, before we include Xilinx on this ?
Thanks

luyong6, 2018-01-13:
Where is Vivado_HLS 2017.4 installed? I just see 2016.4 and 2017.2 installed under ......./fpga/xilinx/Vivado_HLS

ThomasFuchs, 2018-01-14:
Vivado_HLS is now part of the "normal" Vivado installation and can be started with vivado_hls.

luyong6, 2018-01-15:
Got it now. The issue is caused by new HLS doesn't recognize a data structure in C. I have to split the header file to "action_xxxxxx.H" for *hardware action* and "action_xxxxxxx.h" for *software action* and "action_xxxxx_common.h" for *both of them*. Fixing now.

commit 3cc486dc (2018-01-16): Fix #578 HLS intersect: Split header files to sw-only and common (#579)

* HLS intersect: Split header files to sw-only and common (for both)

* fix typo

Signed by Lu Yong (luyong@cn.ibm.com)
closed 2018-01-16 by luyong6