open-power/snap #586: tul2 hang with N250S in reset after flash programming

joergkayser, 2018-01-17:
---- Wed Jan 17 13:19:46 CET 2018---- Executing: ./actions/scripts/snap_jenkins.sh -D SNAP_IMG_TEST_matrix-ACT-hdl_nvme_example.BRAM.NVME-CRD-N250S-99-origin-master -F /opt/fpga/snap/N250S/hdl_nvme_example.BRAM.NVME/fw_2018_0117_1146_NVME_BRAM_N250S_19.bin -A N250S
Using Accel : N250S
Using Image : /opt/fpga/snap/N250S/hdl_nvme_example.BRAM.NVME/fw_2018_0117_1146_NVME_BRAM_N250S_19.bin
Source PATH and LD_LIBRARY_PATH
Flash and test Accel: N250S using: /opt/fpga/snap/N250S/hdl_nvme_example.BRAM.NVME/fw_2018_0117_1146_NVME_BRAM_N250S_19.bin
---> Test Image# 0 File: /opt/fpga/snap/N250S/hdl_nvme_example.BRAM.NVME/fw_2018_0117_1146_NVME_BRAM_N250S_19.bin on N250S
UPDATING Accel: N250S[0] with Image: /opt/fpga/snap/N250S/hdl_nvme_example.BRAM.NVME/fw_2018_0117_1146_NVME_BRAM_N250S_19.bin
Current date: Wed Jan 17 13:19:46 CET 2018
#       Card                           Flashed                       by                   Image
card0    NallatechKU60 Xilinx          Wed Jan 17 11:20:48 CET 2018  fpga                 /opt/fpga/snap/N250S/hdl_nvme_example.BRAM.NVME/fw_2018_0117_1002_NVME_BRAM_N250S_10.bin
card1    AlphaDataKU60 Xilinx          Wed Jan 17 12:43:30 CET 2018  fpga                 /opt/fpga/snap/ADKU3/hdl_example.BRAM/fw_2018_0117_1140_BRAM_ADKU3_6.bin
Continue to flash /opt/fpga/snap/N250S/hdl_nvme_example.BRAM.NVME/fw_2018_0117_1146_NVME_BRAM_N250S_19.bin to card0
Device ID: 0477
Vendor ID: 1014
  VSEC Length/VSEC Rev/VSEC ID: 0x08001280
    Version 0.12
Programming User Partition with /opt/fpga/snap/N250S/hdl_nvme_example.BRAM.NVME/fw_2018_0117_1146_NVME_BRAM_N250S_19.bin
  Program ->  for Size: 56 in blocks (32K Words or 128K Bytes)
Erasing Flash..........
Programming Flash
Writing Block:  0  1  2  3  4  5  6  7  8  9  10  11  12  13  14  15  16  17  18  19  20  21  22  23  24  25  26  27  28  29  30  31  32  33  34  35  36  37  38  39  40  41  42  43  44  45  46  47  48  49  50  51  52  53  54  55  56 
Port not ready 9347160 times
Verifying Flash
Reading Block:  0  1  2  3  4  5  6  7  8  9  10  11  12  13  14  15  16  17  18  19  20  21  22  23  24  25  26  27  28  29  30  31  32  33  34  35  36  37  38  39  40  41  42  43  44  45  46  47  48  49  50  51  52  53  54  55  56 
Erase Time:   63 seconds
Program Time: 28 seconds
Verify Time:  7 seconds
Total Time:   98 seconds
Preparing to reset card
Resetting card

Frank did a reboot, this did NOT bring back the N250S card. Only a power cycle helped to bring both cards back, C0=FGT and C1=KU3

kenhill, 2018-01-17:
This sounds similar to the perst issue being addressed with the new PSL checkpoints. The fix is included in checkpoints with PSL revision level 5. 

This command can verify the revision level on the adapter:
`sudo setpci -s <pci-slot> 40C.B `

joergkayser, 2018-01-17:
not sure, if the machine is really ready to test... Jenkins still hangs, and I don't want to just cancel all IMG_TEST jobs... wait for Eberhard to check status

ThomasFuchs, 2018-01-17:
This should be a image with the new PSL checkpoint... wait for Eberhard to check

ibm-genwqe, 2018-01-18:
I did Check with:
"cat /sys/class/cxl/card0/psl_revision" for Card 0
We do get 12293 which is 05 30. The 05 is the value from offset 40C and 30 is from 40D.
Both Cards on are reporting PSL Version 05.
Previous PSL was "16386" which is 02 40.

joergkayser, 2018-01-26:
Machine works fine again, closing

closed 2018-01-26 by joergkayser