open-power/snap #595: SR10425602: SDRAM BITSLICE placing problem with 2017.4

ThomasFuchs, 2018-01-26:
If I try to build a SNAP image with SDRAM in vivado2017.4 I see placer errors.
The same flow / design works perfectly with vivado 2016.4

I have made a git branch that allows you to reproduce the problem, independent from the SNAP master branch.

git clone https://github.com/open-power/snap.git
git checkout fde63acec0ab75a8d0334180e91ff4cc20be03b5

Error:
Phase 1.2 IO Placement/ Clock Placement/ Build Placer Device ERROR: [Place 30-689] Failed to place BITSLICE_CONTROL cell a0/ddr3sdram_bank0/inst/u_ddr3_mem_intfc/u_mig_ddr3_phy/inst/generate_block1.u_ddr_xiphy/byte_num[0].xiphy_byte_wrapper.u_xiphy_byte_w

ThomasFuchs, 2018-02-28:
From Xilinx:
After been in contact with Development regarding this particular, it seems that it is related to existing CRs, CR-988476 and CR-986318, which were related to an issue with the Pblock algorithm that checks for containment. These have been fixed in 2018.1.

closed 2020-05-15 by bmesnet

Rabbitwow, 2023-08-22:
Hi,
Recently, I'm working on high speed selectio wizard with activated four rx diff pair. And I met this problem also. Is there have any solution for this issues? Please comment if has any idea. Thanks so much. 
W