open-power/snap #596: SR10425625: NVME, empty axi_pcie3_0_0_pcie3_ip.v

ThomasFuchs, 2018-01-26:
NVME create_framework.tcl fails 

I have made a git branch that allows you to reproduce the problem, independent from the SNAP master branch.

git clone https://github.com/open-power/snap.git
git checkout 2eab42575353b58670791c936265aafc1423d12d

ThomasFuchs, 2018-02-05:
I have found a solution that allows me to let the target language in VHDL, but use Verilog files in my simulation. The SNAP project are using two TCL scripts that creates the Vivado project. The NVME and the framework TCL. The target language in the framework TCL is VHDL und unset in the NVME TCL. The root cause of the problem was, that I created the output products for the NVME design with the framework TCL. My solution is now to set the language in the NVME TCL to Verilog and create the output product inside this script instead of the framework script. My tests are currently running and I hope that it works.

boekholt, 2018-02-13:
Reopening, since this fix does break NVMe link initialization in simulation.

ThomasFuchs, 2018-03-02:
NVME simulation runs fine with 2017.4 and 2016.4 in the current master branch

closed 2018-03-02 by ThomasFuchs