open-power/snap #603: "make snap_config" exit without saving

acastellane, 2018-02-01:
Hi,
when you leave the SNAP configuration menu without saving, you don't expect the configuration to change anything.
Can we suppress any action when leaving without saving ?
It can be useful when you want to double check after an image is being build.
Thanks
AC.

joergkayser, 2018-02-01:
Hi @acastellane, this is already true today.
The timestamp of the config files may be changed/touched earlier, but the content of the config files only changes with saving.

acastellane, 2018-02-02:
OK let be more precise, I make image today evening, tomorrow I come back, want to recall my settings with a nice looking menu I'm used to (no manual file edition please), exit no save, I loose my simulation data and my built file. no save => no action :=)

boekholt, 2018-02-02:
I'll look into saving the original config file and compare it to the generated file after execution of kconfig.
In case they do not differ the clean process will be skipped.

boekholt, 2018-02-14:
Pull request #624 got just merged. This pull request is addressing this issue.

@acastellane please close this issue if you are ok with the solution.

Thanks

acastellane, 2018-02-14:
What you propose is a good trade off, some snap files are "refreshed" (same content with a new file date) but existing solutions are not erased.
It allows to use the SNAP "kconfig" menu to double check a configuration after a while.
Thanks

closed 2018-02-14 by acastellane