open-power/snap #607: xsim silulation : impossible to display waveforms of signals outside the top level

bmesnet, 2018-02-01:
When opening the xsim simulator, multiples warnings as the following one:
> WARNING: Simulation object /top/a0/snap_core_i/dma/read_ctrl_reg/cl_calc_v was not traceable in the design for the following reason:                                                                                            
Vivado Simulator does not yet support tracing of VHDL variables. 

It is confirmed that none of the waveform of signals outside the top can be displayed

ThomasFuchs, 2018-02-01:
Hello Bruno,
a VHDL **variables** could not be displayed in XSIM nor in IES. This works as expected. Please do not mix VHDL **variables** with VHDL **signals**!!

bmesnet, 2018-02-02:
ok Thomas agree. My apologize! you are perfectly right, please forget about the warning messages related to variables, BUT......i'm still not able to display a waveform of any **signal** if they are not at the top level with xsim while i can easily do it with irun.
For example : no signal from DMA core is displayable with xsim while it works ok with irun
Is there a setting i'm missing ?

joergkayser, 2018-02-06:
I built an xsim model in Vivado 2017.4 with KU3,hdl_example,DRAM and ran snap_maint and some small app.  I was able to see signals from different units (jobmgr, dma, ...) in the waveform viewer, when running 'xsim top.wdb -gui' from hardware/sim/xsim/latest. This is the directory, where top.wdb resides.
So I cannot reproduce your problem.

There is a core dump, which is created, after I finish xsim. This is being analyzed separately, and there is currently no information, that this is related to your missing signals.

joergkayser, 2018-02-06:
SR10426813 written to Xilinx to get guidance for the "real-time waveform update" feature.

bmesnet, 2018-02-06:
Thanks to sharing the way we work, we understood the issue. 
There is no issue to display waveforms if the **xsim gui** is opened once the simulation completed.
All the tests done  to display waveforms interactively during the simulation, to reload the top.wdb file (which size is increasing in size) drives to the same "no waveforms values displayed" keeping a 6ps width displayed screen.
Closing the xsim after simulation is completed, displays the waveform ok

joergkayser, 2018-02-19:
Status update Feb16 from Xilinx for this request:
I don't believe it is possible to have two instances of Xsim read the same database concurrently, but I am still awaiting a definite answer. I'll push again this morning. It may be something that could be included in a later release of the tools. As soon as I receive a reply either way I will be in touch.

This means: run_sim opens one xsim session to simulate, and the user runs another xsim session to look into the waveform. As a circumvention I could look into adding an option to run_sim, where the waveform GUI is opened together with simulation start, so that no second xsim instance is needed.
@bmesnet  Do you want such an option ?

bmesnet, 2018-02-19:
Well, being able to display dynamically the data that are being generated is what irun does. I guess they are indeed able to open 2 instances and manage that. 
But this is right. If the issue with xsim is that 2 instances may not be able to run the same waveform it may be just the way we use it. We need to try that differently and see what we can do
But if xsim could work as irun, meaning able to look to waveforms at any time even after having started a simulation, this could be a great option.

bmesnet, 2018-02-22:
We use `~snap/hardware/sim/run_sim` script to run the xsim simulator. Is there a way to add in that script the gui option to display the waveform during the simulation ?

joergkayser, 2018-02-22:
I tried in run_sim to pass the argument -gui to xsim, but in this case the Vivado GUI is opened for the simulator, it opens the IP socket for PSLSE, but doesn't pass control back to run_sim to start PSLSE.
This means, no xterm appears, in which we can pass our applications.
No easy solution, on how to start the batch execution AND start the GUI

jsvogt, 2018-03-01:
Feedback from Xilinx: You can't start the waveform GUI after starting the simulation. Xilinx has no plan to get this fixed. @bmesnet Please close

bmesnet, 2018-03-01:
ok good to know. Thx. So the only solution as of today, is then to open the simulator to look to the waveform **after** the simulation is completed and exited.

closed 2018-03-01 by bmesnet