open-power/snap #612: Sim error in snap_example on N250SP with DRAM

joergkayser, 2018-02-05:
When running N250SP with hdl_example DRAM, we get errors from snap_example and snap_example_set. This is with SNAP=master, PSLSE=capi2, Vivado 2017.4, sim=irun
Some test run fine like
snap_example -a1
snap_example -I -a2 -S1 -B0 -A256 -t600
snap_example -a2 -S0 -B2 -A4096 -t400
snap_example -a2 -S0 -B24 -A4096 -t400
snap_example -a2 -S1 -B0 -A4096 -t800
snap_example -a2 -S1 -B2 -A4096 -t800
snap_example -a2 -S1 -B24 -A4096 -t800
snap_example -a2 -S4 -B2 -A4096 -t2000

Waveforms are available in 
/afs/bb/proj/fpga/framework/jkayser/snap_master/hardware/sim/ies/20180205_112927 (snap_example_set)
/afs/bb/proj/fpga/framework/jkayser/snap_master/hardware/sim/ies/20180205_113404 (snap_example -a2 -S4 -B24 -A4096 -t2000 towards the end of the waveform)

joergkayser, 2018-02-06:
The directories have been renamed to keep them for debug:
/afs/bb/proj/fpga/framework/jkayser/snap_612/hardware/sim/ies/20180205_112927 (snap_example_set)
/afs/bb/proj/fpga/framework/jkayser/snap_612/hardware/sim/ies/20180205_113404 /snap_example -a2)

joergkayser, 2018-02-06:
Sorry, something/someone deleted the model andd sim output. Here it is again:
/afs/bb/proj/fpga/framework/jkayser/snap_612/hardware/sim/ies/20180206_170521/ (snap_example_set)

boekholt, 2018-02-19:
We have not yet prepared N250SP for SDRAM use.
I'll add the necessary resources.

boekholt, 2018-02-22:
Please close this issue if pull request #631 does solve it.

joergkayser, 2018-03-15:
no issues with N250SP anymore

closed 2018-03-15 by joergkayser