open-power/snap #615: BUG: Error on NVME Drive 1

ibm-genwqe, 2018-02-06:
Testing snap_example on NVME drive 1 fails;
Failing command: "./actions/hdl_example/sw/snap_example_nvme -C 0 -d 1 -b 1 -v"
NVME Test: Snap Card: /dev/cxl/afu0.0s Timeout: 1 sec NVME Drive: 1 Handle: 0x10025f80010
Host: 0x10025f82000 / 0x10025f83000 DDR: 0x0 / 0x200
    Drive: 1 Size: 0x200 Addr: 0x0 LB: 1 (0x1) BS: 512 (0x200)

        DDR <- HOST 
        NVME <- DDR Error: Timeout while Waiting for Idle 
Exit rc: 1

Tested on:
SNAP FPGA Release: v1.3.0 Distance: 16 GIT: 0xf8e11d1a
SNAP FPGA Build (Y/M/D): 2018/02/05 Time (H:M): 06:54

joergkayser, 2018-02-21:
with the latest branch=nvme_patch_patch we now see the same error in SIM

ThomasFuchs, 2018-02-26:
Fix is in master

ibm-genwqe, 2018-03-01:
Fixed: Thanks
-------------------------------------
Iteration 1 of 1 Drive SSD1
Testing in Polling mode
NVME Test: Snap Card: /dev/cxl/afu0.0s Timeout: 1 sec NVME Drive: 1 Handle: 0x10003280010
Host: 0x10003282000 / 0x10003283000 DDR: 0x0 / 0x200
    Drive: 1 Size: 0x200 Addr: 0x0 LB: 1 (0x1) BS: 512 (0x200)

        DDR <- HOST 
        NVME <- DDR 
        DDR <- NVME 
        HOST <- DDR 
      Compare Buffer Source: 0x10003282000 <-> Destination: 0x10003283000  RC: 0

closed 2018-03-01 by ibm-genwqe