open-power/snap #628: HLS: HLS HashJoin Says BUS Error

ibm-genwqe, 2018-02-15:
Image was:
/opt/fpga/snap/ADKU3/hls_hashjoin/fw_2018_0215_0958_noSDRAM_ADKU3_28.bin
SNAP on ADKU3 Card, NVME disabled, 0 MB SRAM available.
SNAP FPGA Release: v1.3.2 Distance: 5 GIT: 0x6b5679a0
SNAP FPGA Build (Y/M/D): 2018/02/15 Time (H:M): 09:58
SNAP FPGA CIR Master: 1 My ID: 0
SNAP FPGA Up Time: 1 sec
   0 Max AT: 1 Found AT: 0x10141002 --> Assign Short AT: 0
     0     0x10141002     0x00000021  IBM HLS HashJoin
CARD VERSION
[00000000] 010302056b5679a0
[00000008] 0000201802150958

Doing snap_hashjoin ... 
  1 entries for T2 ... ./actions/hls_hashjoin/tests/test_0x10141002.sh: line 85: 31540 Bus error               (core dumped) snap_hashjoin -C1 -T 1 -v >> snap_hashjoin.log 2>&1

bmesnet, 2018-02-21:
Good shot. Duplication of a snap_detach call in the snap_hashjoin application
Code corrected in branch SNAP_documentation. Pull request follows

bmesnet, 2018-02-21:
Pull request #638

bmesnet, 2018-02-22:
@ibm-genwqe can you check that this issue is now corrected. It is already pushed to master since this bug correction was pushed at the same time than pushing general documentation which was asked by more and more people.

ibm-genwqe, 2018-03-01:
HLS Hash join did pass Jenkins Testing
-------------
 8088 entries for T2 ... ok
  123123 entries for T2 ... ok
Test OK
RUN: ./actions/hls_hashjoin/tests/test_0x10141002.sh -C 1 -d NORMAL on ADKU3[1] Done RC=0
Wed Feb 28 13:23:05 CET 2018 Image Test on Accel: ADKU3 was executed 1 time(s)
Wed Feb 28 13:23:06 CET 2018

closed 2018-03-01 by ibm-genwqe