open-power/snap #630: HLS_memcopy hangs after timeout

joergkayser, 2018-02-20:
when running hls_memcopy on any card, the application snap_memcopy hangs, and even after the application timeout of 60sec the app still hangs, until Jenkins kills the session (after 300min or more).

The SNAP_TRACE shows the  60sec timeout and 2 more commands, before it hangs
err: job execution -6: Timer expired!
D snap_detach_action Enter
D hw_detach_action Enter Action: 0x10141000 Base: f000 timeout: 60 sec Seq: 0xf002
      got end of exec. time
R   hw_snap_mmio_read32(0x21ad050, f000, 1) 0
R   hw_snap_mmio_write64(0x21ad050, 1010, 4)

SNAP branch=nvme_patch_patch (master fails as well)
PSLSE branch=capi2 (export PSLVER=8)
APP log see /afs/bb/proj/fpga/framework/jkayser/snap_hls_memcopy/hardware/sim/ies/20180220_144901/testlist1.log
Waveform is in the same directory

joergkayser, 2018-02-20:
The directory /afs/bb/proj/fpga/framework/jkayser/snap_hls_memcopy/hardware/sim/ies/20180220_151441 is better for analysis, as the waveform is complete from start to end. It was built with PSLSE v3.1 and shows the same behavior.

joergkayser, 2018-02-20:
The error must have been introduced between Jan31 (PortMap works) and Feb2 (SR10425625 fails). Since then all Jenkins runs show "aborted", which is the Jenkins answer to the maximum timeout limit.
SR10425625 switched the target_language from VERILOG to VHDL. Is this the root cause ?

boekholt, 2018-02-28:
@bmesnet 
Hi Bruno, Thomas and me looked into this and, it looks like this is an hls_memcopy problem. We see that the read operation from host memory completes successfully. Then the action wants to send data back to host memory. It raises AWVALID on the AXI interface and gets AWREADY as response. Then it should transfer the data but the data does not show up at WDATA and WVALID does not raise.

Given the timeframe indicated by Joerg it looks like the problem got introduced with pull request #600 (which has a misleading title because it also contains functional changes in hls_memcopy.cpp).

I am assigning this issue to you.
Thanks
Sven

bmesnet, 2018-03-01:
This is the hls_memcpy issue I already face earlier when number of bytes are lower than 1 word. Trying to isolate the issue and propose a patch asap.

boekholt, 2018-03-02:
#653 does solve the issue.

closed 2018-03-02 by boekholt