open-power/snap #632: N250SP+SDRAM: snap_example_ddr throws core dump

joergkayser, 2018-02-20:
SNAP branch: sdram4n250sp
core dump info: snap_example_ddr: double free or corruption (out): 0x0000000001a86000
gdb backtrace:
Core was generated by `snap_example_ddr -i1 -b448 -s2475 -e2923 -t412'.
Program terminated with signal 6, Aborted.
#0  0x0000003faec32625 in raise () from /lib64/libc.so.6
(gdb) bt
#0  0x0000003faec32625 in raise () from /lib64/libc.so.6
#1  0x0000003faec33e05 in abort () from /lib64/libc.so.6
#2  0x0000003faec70537 in __libc_message () from /lib64/libc.so.6
#3  0x0000003faec75f4e in malloc_printerr () from /lib64/libc.so.6
#4  0x0000003faec78cf0 in _int_free () from /lib64/libc.so.6
#5  0x00000000004030f2 in main (argc=<optimized out>, argv=<optimized out>) at snap_example_ddr.c:786

core dump can be found in /afs/bb/proj/fpga/framework/CI_ws/SNAP_SIM_matrix_mult/sdram4n250sp/ACT/hdl_example.SDRAM/CRD_VIV/N250SP_17/SIM/irun/snap/hardware/sim/ies/latest

ibm-genwqe, 2018-03-01:
The Size in this Test for CAPI 2.0 on P9 must be a multiple of 128 bytes. Please set -b to n*128 and not 448.

joergkayser, 2018-03-15:
works now. Fix is in branch=Issue_665

closed 2018-03-15 by joergkayser