open-power/snap #637: Multiboot fails after flashing 0-byte user image

jsvogt, 2018-02-21:
When the user image bin file is 0 bytes the flasher will erase the user partition (separate issue for flasher tool). After a reset, the card then does not recover from the factory partition. 
Flashing both partitions with a mcs file again repairs the card.

=> Debug if this is caused by an unreliable `echo 1 > /sys/class/cxl..../reset` or if the Multiboot Fallback mechanism doesn't work right.

joergkayser, 2018-02-21:
The problem was most likely introduced by Jenkins due to an AFS_full condition. The zero_Byte file was xferred to the test station and programmed.
Future Jenkins runs will validate a minimum size before xfer&test and error_exit, if not

jsvogt, 2018-07-11:
The few test results so far weren't consistent, as sometimes a connected vivado_lab instance forced an N250S card into booting from the factory partition.
Meanwhile the snap_bitstream_pre.tcl script is separate for each card, and there are cards with SPI config flash(es).
=> Each card must be tested separately. 
1. Test one of the major CAPI 2.0 cards, such as N250S+, and fix errors there first before testing more cards.

jsvogt, 2018-12-12:
Related: N250S+ card got a PSL9 reset fix that had prevented us from testing this so far.
See https://github.com/open-power/capi2-bsp/pull/17

bmesnet, 2020-05-15:
solved

closed 2020-05-15 by bmesnet