open-power/snap #646: "make software" doesn't read snap_env.sh variables

acastellane, 2018-02-27:
hi,
when we are in XXX/snap and we have XXX/pslse set by default the make software will find the pslse directory even if the variable PSLSE_ROOT is not available (because of the default)
But when variable PSLSE_ROOT is not set AND pslse is not in the default location, it is not found.

A message occurs informing that pslse is not found at XXX/pslse.

Software Makefile content :
_ifndef PSLSE_ROOT
# Environment variable PSLSE_ROOT defined by hardware setup scripts.
# Use default path if PSLSE_ROOT is not defined.
PSLSE_ROOT=$(abspath ../../pslse)
endif_

Proposed solution : in snap/Makefile execute $(snap_env_sh) in software section ?
Regards
AC.

boekholt, 2018-03-01:
Before it may be used, the file `$(snap_env_sh)` needs to be created. In the regular flow this requires that `snap_conifg.sh` does exist. So far, software build was independent of that. We could add the sourcing of `$(snap_env_sh)` directly in the `PSLSE_ROOT` definition section of `software/Makefile` in the case that `PSLSE_ROOT` is not yet defined while `$(snap_env_sh)` is existing - then a check needs to be added if `PSLSE_ROOT` actually got defined. I guess, all this will end up in a quite complicated construct that @fhaverkamp won't like.

boekholt, 2018-03-15:
The solution got merged into master 10 days ago.
I am closing this issue now.
Please reopen if the solution does not meet the requirements.

closed 2018-03-15 by boekholt