open-power/snap #651: X86 vs Power libcxl Library compilation

acastellane, 2018-02-28:
Issue :
Assume you're using the exact same working directory for a project (say nimbix /data located snap directory) and moving from development to deployment, back and forth.
Should you want to use the tools or the snap applications on Power, they need recompilation. Running a "make" will lead you to an error as libcxl library is seen as already compiled in x86 mode.
Assume you don't want to loose your model and image report, etc, you don't want to clean everything.
Then it becomes a bit tricky to selectively clean and recompile only what you need.

Proposed solution : 
When compiling libcxl, 2 different directories could be used for each type of machine as we detect it early in the process. This would simplify the development process.

thanks

boekholt, 2018-02-28:
So, you are using the same directories on Power and on x86?
How do you do that?

acastellane, 2018-02-28:
As mentioned in my issue description with Nimbix, we can share /data folder between instances.
Thanks

fhaverkamp, 2018-07-11:
Im my eyes the required effort does not really justify the benefits we get from this. If you agree, I would like to close this. We might keep multiplatform build in mind when establishing SNAP for OpenCAPI.

acastellane, 2018-07-16:
Proposed another solution : instead of dealing with 2 directories, which looks painful, we could either leave a trace of what is has been compiled for, or check platform consistency and eventually enforce recompilation for the desired platform.
Thanks

fhaverkamp, 2018-07-17:
https://github.com/open-power/snap/tree/SW_add_check_platform

After playing a little bit with how to get the exit codes properly processed in bash and Makefile, here a first idea how it might look like. I wonder if that might have negative side effects ... ;-).
Have a look if that works for you. Feel free to fix it up in the same branch if you discover things
not working right.

acastellane, 2018-07-23:
Successfully tested SW_add_check_platform branch on cloud, it works great. We can now use the same directory to prepare our development work and deploy on POWER systems.
If this is ok, please include the platform test in master.
Thanks

closed 2018-09-13 by acastellane