open-power/snap #652: HLS : memcopy doesn't work with small packets

bmesnet, 2018-03-01:
Old bug refered in #320 but hidden by circumvention used until now.
Need to localize exact case of failure to open an SR to Xilinx
Meanwhile putting back the circumvention which works in all cases:

```
       // Prepare Patch to the issue# - memcopy doesn't handle small packets
        int size_in_words;
        if(size_in_bytes_to_transfer %BPERDW == 0)
                size_in_words = size_in_bytes_to_transfer/BPERDW;
        else
                size_in_words = (size_in_bytes_to_transfer/BPERDW) + 1;
        //end of patch
```

        switch (memory_type) {
        case SNAP_ADDRTYPE_HOST_DRAM:
                // Patch to the issue# - memcopy doesn't handle small packets
                //memcpy((snap_membus_t  *) (dout_gmem + output_address),
                //       buffer, size_in_bytes_to_transfer);

                // Do not insert anything more in this loop to not break the burst
                wb_dout_loop: for (int k=0; k<size_in_words; k++)
                #pragma HLS PIPELINE
                    (dout_gmem + output_address)[k] = buffer[k];
                // end of patch`

bmesnet, 2018-06-20:
More investigation done on the issue:
The problem is clearly identified as using HLS **memcpy** function to **write** a number of bytes **smaller** than the port data width. Problem addressed to Xilinx. 

Our port to DDR or Host memories is always 64Bytes in SNAP. 
Reading for example 1 Byte can be done with a fix value or a parameter. In both cases it works ok:
With SIZE_WORD defined to 512 and xfer_size defined to 1
memcpy(buff, (ap_uint<SIZE_WORD>*)a, 1); 		//this read from the port works ok
memcpy(buff, (ap_uint<SIZE_WORD>*)a, xfer_size); 	//this read from the port works ok

Writing for example 1 Byte can be done with a fix value or a parameter. Only the fixed value works ok
memcpy((ap_uint<SIZE_WORD> *)a, buff, 1);		//this write to the port works ok
memcpy((ap_uint<SIZE_WORD> *)a, buff, xfer_size);	//this write to the port DOESN'T work ok

Problem doesn't happen if the number of bytes read with the memcpy is greater or equal to the port word width!
From what I looked in the VHDL generated by HLS, it seems that when we put a variable instead of a fixed value, the size is strangely shifted to be aligned on the size of the bus!
=> Waiting for Xilinx feedback/SR number

bmesnet, 2018-07-02:
Problem still alive with HLS2018.2 => ticket opened **CR-1006181**

bmesnet, 2018-09-24:
Fix has been identified and corrected by Xilinx. 
Fix will be available in Vivado HLS 2018.3 second part of November

abbasBSC, 2019-05-24:
> More investigation done on the issue:
> The problem is clearly identified as using HLS **memcpy** function to **write** a number of bytes **smaller** than the port data width. Problem addressed to Xilinx.
> 
> Our port to DDR or Host memories is always 64Bytes in SNAP.
> Reading for example 1 Byte can be done with a fix value or a parameter. In both cases it works ok:
> With SIZE_WORD defined to 512 and xfer_size defined to 1
> 
> ```
> memcpy(buff, (ap_uint<SIZE_WORD>*)a, 1); 		//this read from the port works ok
> memcpy(buff, (ap_uint<SIZE_WORD>*)a, xfer_size); 	//this read from the port works ok
> ```
> 
> Writing for example 1 Byte can be done with a fix value or a parameter. Only the fixed value works ok. When using a variable, the size of the transfer is considered as 0 and the transfer never ends.
> 
> ```
> memcpy((ap_uint<SIZE_WORD> *)a, buff, 1);		//this write to the port works ok
> memcpy((ap_uint<SIZE_WORD> *)a, buff, xfer_size);	//this write to the port DOESN'T work ok
> ```
> 
> Problem doesn't happen if the number of bytes read with the memcpy is greater or equal to the port word width!
> From what I looked in the VHDL generated by HLS, it seems that when we put a variable instead of a fixed value, the size is strangely shifted to be aligned on the size of the bus!
> => Waiting for Xilinx feedback/SR number

HI Bruno, hls_memcpy axi_mm data transfer works in burst mode. I believe that the number of burst transfers is defined in HLS headr file: #define MAX_NB_OF_BYTES_READ	(256 * 1024) which is more than 4KB.
I tried to change this number from 64 (port width) to 64x64 (4KB) to see how different burst modes affect execution time. It still do not work for numbers other than 256x1024. it gives same time out error as for small packets in simulation and "err: Unexpected RETC=0!" on power9

bmesnet, 2019-05-24:
Hi Abbas, 
AXI_MM's burst is limited to 4KB, but the HLS memcpy will manage that by it's own meaning that if you ask for more than 4KB, he will spread it in 4KB bursts, but this is transparent for the user.
The port width is fixed and defined in `actions/include/hls_snap.H`. We have defined this as a fixed value to prevent multiple width conversion and latency delay.
If you want to reduce the size of the AXI burst within the hls action, you need to change the value of `max_read_burst_length=64  max_write_burst_length=64` which are defined with the `#pragma HLS INTERFACE` of the bus you intend to play with. By default if you don't specify these  2 variables, AXI burst will be 1KB.
Does it answers your question?

bmesnet, 2019-05-24:
This issue #652 is related to write of small blocks only and is corrected in HLS 2018.3.

abbasBSC, 2019-05-27:
> Hi Abbas,
> AXI_MM's burst is limited to 4KB, but the HLS memcpy will manage that by it's own meaning that if you ask for more than 4KB, he will spread it in 4KB bursts, but this is transparent for [...]

Yes, very clear. Thank you.

bmesnet, 2020-03-05:
Closing issue since checked as fixed and users should now be at a higher vivado release. Ccircumvention has been kept in hls_helloworld example, if needed

closed 2020-03-05 by bmesnet