open-power/snap #656: Support simulation in Modelsim

luyong6, 2018-03-05:
Hi
To make the SNAP more easy to use, do we consider to support Modelsim for simulation? 
Many universities and academic instituions prefer to use Modelsim as their simulation tool and want to use their familiar scripts and facilities.

joergkayser, 2018-03-05:
For this discussion we need some more information:
the free version modelsim or the licensed version questa ? Which version of modelsim ?
And I assume, modelsim is not just the product from Mentor Graphics, but also delivered with some other development suites. We definetely should concentrate on simulators, which are supported by Vivado (currently using version 2017.4)

hsukiang, 2018-03-05:
The university partner that I supported uses the full license for ModelSim. I have been using Xilinx's license from an academic partner to do my testing for Vivado / SDAccel tools. I am currently upgrading my tools to 2017.4 so that I can get acquainted with XSIM.

joergkayser, 2018-03-05:
@hsukiang Which version of Modelsim is your university partner using ? I need to check, if we have licenses and if Vivado supports this version.

hsukiang, 2018-03-06:
@joergkayser They have adopted ModelSim 10.4b recently. Thank you.

joergkayser, 2018-03-06:
The IBM systems installations I know are
Altera Modelsim-ASE versions 13.0, 13.1, 14.02, 15.0, 15.1 (CTE installation dated 2015)
Mentor Modelsim SE versions 5.8c, 6.0a, 6.2h, 6.3g, 6.6b (CTE installation dated 2010)
Mentor Questa version 10.2b (local installation dated 2013)
The Mentor release letter advertises Modelsim SE/PE/DE version 10.6d
Which version stream should we follow ?

hsukiang, 2018-03-07:
We found that the free version (SE) has limitation with the number of executable lines of code. Due to this, we installed the AE edition that requires license. The university uses academic license for that.
In IBM case, we should follow the ASE stream of the latest release installed. The AE and ASE are both "Altera" Edition, with the ASE performance being limited. If we can get ASE working, we should not have issues with AE.

luyong6, 2018-03-08:
Lance said XXXX client routinely uses mentor graphics questa.

joergkayser, 2018-04-04:
@luyong6 @hsukiang please checkout branch=questa_trial and try to use questa on a model of your choice.
I made following observations:
- Vivado 2017.4 checks the questa version, before compiling the libraries. It wants questa 10.6, and this is not available on our site.
- Vivado 2016.4 does not check the questa version, and I tried to compile the simulation libraries with Questa 10.2 and failed with several compile errors and a core dump (questa_sim/linux_x86_64/vlog: corrupted double-linked list: 0x0000000002fa8a30 )
- The free version of Altera Starter Edition (modelsim ASE) is not supported by VIVADO and cannot be selected in the Vivado GUI. If you select 'Modelsim' and point to Altera modelsim ASE instead of the Mentor modelsim SE, the simulator is not found.

In general Questa should be supported with following settings:
- export MODELSIM_DIR=<path to questa_sim>
- export MENTOR_LIBS= <path to Vivado IP compiled with questa, see vivado, compile simlib>
- export PATH=$MODELSIM_DIR/bin:$PATH
- export MGLS_LICENSE_FILE=<your Mentor license>

I was not able to request a version update, as we dont have licenses for this.  The msg I got was:
We have 3 (old V6) modelsim vhdl licenses and then have stopped paying maintenance and using this.  We do not have any modelsim verilog licenses in our global pool.  

Please check with your partners, if the available branch suits their needs, but we cannot provide more support without having Mentor licenses for testing

luyong6, 2018-04-11:
Hi Joerg, Thank you very much. I will ask a university partner to test it.

joergkayser, 2018-06-07:
modelsim + questa + xcelium are supported since a while. Closing this ticket. Please note, that we do not use modelsim ourselves. Please reopen or open a new ticket, if any of this features are not good enough for external use.

closed 2018-06-07 by joergkayser