open-power/snap #658: hls_search: bad timing

ThomasFuchs, 2018-03-05:
Hi,
I am no longer able to build hls_search example images, because the timing is really bad. Please have a look:

A complete FPGA bitstream build got kicked off.
This might take more than an hour depending on the machine used
The process may be terminated by pressing <CTRL>-C at any time.
After termination it can be restarted later.

                        patching SNAP version and build date registers
INFO: [Common 17-1239] XILINX_LOCAL_USER_DATA is set to 'no'.
                        open framework project                                      17:31:47 Fri Mar 02 2018
                        start synthesis         with directive: Default             17:32:01 Fri Mar 02 2018
                        start opt_design        with directive: Explore             17:56:50 Fri Mar 02 2018
                        reload opt_desing DCP                                       18:08:49 Fri Mar 02 2018
                        Prevent placing inside  PSL                                 18:11:26 Fri Mar 02 2018
                        start place_design      with directive: Explore             18:11:26 Fri Mar 02 2018
                        start phys_opt_design   with directive: Explore             18:30:51 Fri Mar 02 2018
                        start route_design      with directive: Explore             18:48:17 Fri Mar 02 2018
                        start opt_routed_design with directive: Explore             00:36:34 Sat Mar 03 2018
                        generating reports                                          03:38:57 Sat Mar 03 2018
                        Timing (WNS)            -2354 ps                                                  
                                                ERROR: TIMING FAILED

luyong6, 2018-03-05:
All cards? 
Which card for this case? 
This WNS is a little terrible. Something must be wrong.

bmesnet, 2018-03-15:
@ThomasFuchs I'm working on it on a separate branch "HLS_search_update". I did some cleaning and have some good results. I will tell you when this can be tested Not everything is pull up yet in github

bmesnet, 2018-03-15:
@ThomasFuchs  #680 pull request should solve your timing issue. Let jenkins see if that's ok on this branch

bmesnet, 2018-03-16:
Pullrequest#680 merged to master

bmesnet, 2018-03-16:
Issue #682 found after merge is independent of this timing issue

ThomasFuchs, 2018-03-20:
Timing looks pretty good for hls_search now

closed 2018-03-20 by ThomasFuchs