open-power/snap #665: N250SP+hdl_example: intermittend error with snap_example -a6 -S20 -B20 -A4096

joergkayser, 2018-03-06:
see
/afs/bb/proj/fpga/framework/CI_ws/SNAP_SIM_matrix/n250sp_testlist/ACT/hdl_example.SDRAM/CRD/N250SP/SIM/irun/VIV/2017.4/snap/hardware/sim

joergkayser, 2018-03-06:
also fails within 5 loops in
snap_example -a6 -S20 -A4096
snap_example -a6 -S10 -A1024
snap_example -a6 -S10 -A128
snap_example -a6 -S4   -A128
and within 10 loops in
snap_example -a6 -S2 -A128
snap_example -a6 -S1 -A4096

joergkayser, 2018-03-06:
00000000: 00 00 00 00 ff ff ff ff 08 00 00 00 f7 ff ff ff | ................
 00000010: 10 00 00 00 ef ff ff ff 18 00 00 00 e7 ff ff ff | ................
 00000020: 20 00 00 00 df ff ff ff 28 00 00 00 d7 ff ff ff | ................
 00000030: 30 00 00 00 cf ff ff ff 38 00 00 00 c7 ff ff ff | 0.......8.......
 00000040: 40 00 00 00 bf ff ff ff 48 00 00 00 b7 ff ff ff | ........H.......
 00000050: 50 00 00 00 af ff ff ff 58 00 00 00 a7 ff ff ff | P.......X.......
 00000060: 60 00 00 00 9f ff ff ff 68 00 00 00 97 ff ff ff | ........h.......
 00000070: 70 00 00 00 8f ff ff ff 78 00 00 00 87 ff ff ff | p.......x.......
 00000080: 80 **01** 00 00 7f fe ff ff 88 01 00 00 77 fe ff ff | ............w...
 00000090: 90 **01** 00 00 6f fe ff ff 98 01 00 00 67 fe ff ff | ....o.......g...
 000000a0: a0 01 00 00 5f fe ff ff a8 01 00 00 57 fe ff ff | ............W...
 000000b0: b0 01 00 00 4f fe ff ff b8 01 00 00 47 fe ff ff | ....O.......G...
 000000c0: c0 01 00 00 3f fe ff ff c8 01 00 00 37 fe ff ff | ............7...
 000000d0: d0 01 00 00 2f fe ff ff d8 01 00 00 27 fe ff ff | ................
 000000e0: e0 01 00 00 1f fe ff ff e8 01 00 00 17 fe ff ff | ................
 000000f0: f0 01 00 00 0f fe ff ff f8 01 00 00 07 fe ff ff | ................
 00000100: 00 02 00 00 ff fd ff ff 08 02 00 00 f7 fd ff ff | ................
 00000110: 10 02 00 00 ef fd ff ff 18 02 00 00 e7 fd ff ff | ................
 00000120: 20 02 00 00 df fd ff ff 28 02 00 00 d7 fd ff ff | ................
 00000130: 30 02 00 00 cf fd ff ff 38 02 00 00 c7 fd ff ff | 0.......8.......
 00000140: 40 02 00 00 bf fd ff ff 48 02 00 00 b7 fd ff ff | ........H.......
 00000150: 50 02 00 00 af fd ff ff 58 02 00 00 a7 fd ff ff | P.......X.......
 00000160: 60 02 00 00 9f fd ff ff 68 02 00 00 97 fd ff ff | ........h.......
 00000170: 70 02 00 00 8f fd ff ff 78 02 00 00 87 fd ff ff | p.......x.......
 00000180: 80 01 00 00 7f fe ff ff 88 01 00 00 77 fe ff ff | ............w...
 00000190: 90 01 00 00 6f fe ff ff 98 01 00 00 67 fe ff ff | ....o.......g...
 000001a0: a0 01 00 00 5f fe ff ff a8 01 00 00 57 fe ff ff | ............W...
 000001b0: b0 01 00 00 4f fe ff ff b8 01 00 00 47 fe ff ff | ....O.......G...
 000001c0: c0 01 00 00 3f fe ff ff c8 01 00 00 37 fe ff ff | ............7...
 000001d0: d0 01 00 00 2f fe ff ff d8 01 00 00 27 fe ff ff | ................
 000001e0: e0 01 00 00 1f fe ff ff e8 01 00 00 17 fe ff ff | ................
 000001f0: f0 01 00 00 0f fe ff ff f8 01 00 00 07 fe ff ff | ................
 00000200: 00 02 00 00 ff fd ff ff 08 02 00 00 f7 fd ff ff | ................
 00000210: 10 02 00 00 ef fd ff ff 18 02 00 00 e7 fd ff ff | ................
 00000220: 20 02 00 00 df fd ff ff 28 02 00 00 d7 fd ff ff | ................
 00000230: 30 02 00 00 cf fd ff ff 38 02 00 00 c7 fd ff ff | 0.......8.......
 00000240: 40 02 00 00 bf fd ff ff 48 02 00 00 b7 fd ff ff | ........H.......
 00000250: 50 02 00 00 af fd ff ff 58 02 00 00 a7 fd ff ff | P.......X.......
 00000260: 60 02 00 00 9f fd ff ff 68 02 00 00 97 fd ff ff | ........h.......
 00000270: 70 02 00 00 8f fd ff ff 78 02 00 00 87 fd ff ff | p.......x.......
 00000280: 80 02 00 00 7f fd ff ff 88 02 00 00 77 fd ff ff | ............w...
 00000290: 90 02 00 00 6f fd ff ff 98 02 00 00 67 fd ff ff | ....o.......g...
 000002a0: a0 02 00 00 5f fd ff ff a8 02 00 00 57 fd ff ff | ............W...
 000002b0: b0 02 00 00 4f fd ff ff b8 02 00 00 47 fd ff ff | ....O.......G...
 000002c0: c0 02 00 00 3f fd ff ff c8 02 00 00 37 fd ff ff | ............7...
 000002d0: d0 02 00 00 2f fd ff ff d8 02 00 00 27 fd ff ff | ................
 000002e0: e0 02 00 00 1f fd ff ff e8 02 00 00 17 fd ff ff | ................
 000002f0: f0 02 00 00 0f fd ff ff f8 02 00 00 07 fd ff ff | ................

joergkayser, 2018-03-06:
snap_example -a2 -S1 -A128 fails as well, similar symptom (second byte at buffer begin is wrong)

joergkayser, 2018-03-07:
The yesterdays data was deleted by mistake, and the recreate today was not succesful with very small data or just host access (-a2 -S1)
However, I was able to recreate it in loop=4 with
for i in {1..5};do echo "loop $i";snap_example -a6 -S8 -B2 -A128 -t200 -v||break;done

see  /afs/bb/proj/fpga/framework/CI_ws/SNAP_SIM_matrix/n250sp_testlist/ACT/hdl_example.SDRAM/CRD/N250SP/SIM/irun/VIV/2017.4/snap/hardware/sim/ies/20180307_142714

joergkayser, 2018-03-07:
the error case was copied from CI_ws to a seperate debug directory and fails in the first loop
see /afs/bb/proj/fpga/framework/debug/xx/snap/hardware/sim/ies/20180307_171234

joergkayser, 2018-03-08:
recreated with stim.log and mismatch memcopy map in 
/afs/bb/proj/fpga/framework/debug/xx/snap/hardware/sim/ies/20180308_122829
cmd= for i in {1..5};do echo "loop $i";snap_example -a2 -S1 -B8 -A128 -t200 -vvvv||break;done

ThomasFuchs, 2018-03-09:
This was a problem with the write buffer read address handling in dma_fsm_q. Fix is in #672

joergkayser, 2018-03-12:
problem still exixsts with the latest code from #7383b9a7 ("Config: Reverting incorrect change of FPGACHIP for N250SP" )
Model copied with ../common/copy_debug
from  /afs/bb/proj/fpga/framework/CI_ws/SNAP_SIM_matrix/Issue_665/ACT/hdl_example.SDRAM/CRD/N250SP/SIM/irun/VIV/2017.4 
to  ../debug/issue_665_part2
and reproduced with AET and snap_example -a2 -S20 -B28 -A128
see /afs/bb/proj/fpga/framework/debug/issue_665_part2/snap/hardware/sim/ies/20180312_114817/stim.log

joergkayser, 2018-03-13:
previous problem (DMA ?) doesnt exist anymore, and we now fail at a later point in the testlist with snap_example_set -H -b0 -s128 -p128 -t306 -vvv
This is due to data being moved by 16B and xferred via DMA. Investigations ongoing, why this breaks and how to rewrite the test to make it pass.

joergkayser, 2018-03-14:
After some SW tweaking we still fail in snap_example_set -H -b0 -s128 -p128 -t306 -vvv
with
Start Memset Test. Timeout: 306 sec Device: INFO:Connecting to host 'hdcl070.boeblingen.de.ibm.com' port 16384
D   hw_snap_card_alloc_dev: errinfo_size: 1 VendorID: 1014 DeviceID: 632
D hw_snap_card_alloc_dev Exit 0x1183010 OK Context: 0 Master: 0
/dev/cxl/afu0.0s
D   hw_card_ioctl Get MEM: 8192 MB
Allocate Host Buffer at 0x1185000 Size: 320 Bytes
Fill 128 Bytes from 0 to 127 with Pattern: 0x80 FPGA RAM: 8192 MB
D hw_attach_action Enter Action: 0x10140000 Old Action: ffffffff Flags: 0x0 Base: 0 timeout: 1530 sec Seq: 0
R   hw_snap_mmio_read64(0x1183010, 18, 100) 0
R   hw_snap_mmio_read64(0x1183010, 100, 10140000) 0
R   hw_snap_mmio_write64(0x1183010, 1000, f000000000000001)
R   hw_snap_mmio_write64(0x1183010, 1010, f000000000000001)
R   hw_snap_mmio_read64(0x1183010, 1008, f000f00000000081) 0
R   hw_snap_mmio_read64(0x1183010, 1008, f000f000000000c3) 0
D hw_attach_action Exit rc: 0 Action: 0x1183010 Base: 0xf000
[1/1] Start Memset  Attach 10140000
R   hw_snap_mmio_write32(0x1183010, f030, 8008)
R   hw_snap_mmio_write32(0x1183010, f034, 1185040)
R   hw_snap_mmio_write32(0x1183010, f038, 0)
R   hw_snap_mmio_write32(0x1183010, f03c, 11850bf)
R   hw_snap_mmio_write32(0x1183010, f040, 0)
D snap_action_start: START Action 0x10140000 Flags 0
R   hw_snap_mmio_write32(0x1183010, f000, 1)
[Host] Set: 127 Bytes with Pattern: 0x80 R   hw_snap_mmio_read32(0x1183010, f000, 8) 0
R   hw_snap_mmio_read32(0x1183010, f000, c) 0
 T = 286 msec
Check 128 bytes in Buffer Start at: 0x40 for Pattern: 0x80
 00000000: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff | ................
 00000010: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff | ................
 00000020: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff | ................
 00000030: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff | ................
 00000040: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff | ................
 00000050: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff | ................
 00000060: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff | ................
 00000070: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff | ................
 00000080: 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 | ................
 00000090: 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 | ................
 000000a0: 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 | ................
 000000b0: 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 | ................
 000000c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff | ................
 000000d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff | ................
 000000e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff | ................
 000000f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff | ................

Error@: 0x40 Expect: 0x80 Read: 0xff
Error@: 0x41 Expect: 0x80 Read: 0xff

see /afs/bb/proj/fpga/framework/debug/issue_665_part3/snap/hardware/sim/ies/20180313_165230

joergkayser, 2018-03-15:
works fine now, included in branch Issue_665

closed 2018-03-15 by joergkayser