open-power/snap #682: HLS_search_update: 2nd call to hls_search fails with timeout on all configurations

joergkayser, 2018-03-16:
action was ok until March 15, error was introduced with branch=HLS_search_update and now seen in master as well
AET see  /afs/bb/proj/fpga/framework/debug/issues_search_irun/snap/hardware/sim/ies/20180316_141612/

bmesnet, 2018-03-16:
This problem is due to a change I did in the application to **enable the interrupt mode by default** (like all other examples - changing the -X option by -N option).
The way IRQ are coded in the code is identical than the way it is done on other examples (hls_memcopy for example), the only specific thing in this hls_search is that we are using `snap_queue_alloc` instead of typical `snap_attach_action` used in other examples. Are the IRQ defined/working the same way on **queue** mode than on **attach** mode ? Previous jenkins tests was not testing interrupt mode but only polling mode. Problem is not seen with polling mode (-N option)

Test run on latest master code (simulation or FPGA execution)
echo "Hello SNAP world. This is my first CAPI SNAP experience. It's real fun!" > /tmp/t1
snap_search -t5000 -m1 -E2 -i /tmp/t1 -p SNAP       **<= works ok**
snap_search -t5000 -m1 -E2 -i /tmp/t1 -p SNAP       **<= fails**
using polling mode (adding -N argument) **works ok**

hls_search example execution is made of 2 steps : Step1 for a memcopy and Step3 for a search
First call always work ok
2nd call always fail when starting Step1 with the following timeout error:

> hdclf009 20180316_153233$ SNAP_TRACE=0xF snap_search -t5000 -m1 -E2 -i /tmp/t1 -p SNAP
D hw_snap_card_alloc_dev Enter /dev/cxl/afu0.0s
INFO:Connecting to host 'hdclf009.boeblingen.de.ibm.com' port 16384
D   hw_snap_card_alloc_dev: errinfo_size: 1 VendorID: 1014 DeviceID: 632
D hw_snap_card_alloc_dev Exit 0x212a090 OK Context: 0 Master: 0
...................................................
Start Step1 (Copy source data from Host to DDR) ...
INITIALIZATION : move 71 bytes from Host mem to DDR
D **hw_attach_action Enter** Action: 0x10141003 Old Action: ffffffff Flags: 0x10001 Base: 0 timeout: 60 sec Seq: 0
R   hw_snap_mmio_read64(0x212a090, 18, 100) 0
R   hw_snap_mmio_read64(0x212a090, 100, 10141003) 0
R   hw_snap_mmio_write64(0x212a090, **1000**, f000000000000005)
R   hw_snap_mmio_write64(0x212a090, **1010**, f000000000000001)
D   hw_wait_irq: Enter fd: 4 Flags: 0x10001 Expect irq: 2 Timeout: 60 sec
D     Timeout......
D   hw_wait_irq: Exit fd: 4 rc: 16
D **hw_attach_action Exit** rc: 16 Action: (nil) Base: 0x0
D **snap_sync_execute_job: Error Can not attach to Action 0x10141003**
err: job execution -8: Timer expired!
INFO:detach response from from pslse

The first run went ok (see below). It seems that in 2nd call/Step3 there are missing writes to address 1000. Could it be the reason ?

> hdclf009 20180316_153233$ SNAP_TRACE=0xF snap_search -t5000 -m1 -E2 -i /tmp/t1 -p SNAP
D **hw_snap_card_alloc_dev Enter** /dev/cxl/afu0.0s
INFO:Connecting to host 'hdclf009.boeblingen.de.ibm.com' port 16384
D   hw_snap_card_alloc_dev: errinfo_size: 1 VendorID: 1014 DeviceID: 632
D **hw_snap_card_alloc_dev Exit** 0x1e57090 OK Context: 0 Master: 0
...................................................
Start Step1 (Copy source data from Host to DDR) ...
INITIALIZATION : move 71 bytes from Host mem to DDR
D **hw_attach_action Enter Action**: 0x10141003 Old Action: ffffffff Flags: 0x10001 Base: 0 timeout: 60 sec Seq: 0
R   hw_snap_mmio_read64(0x1e57090, 18, 100) 0
R   hw_snap_mmio_read64(0x1e57090, 100, 10141003) 0
R   hw_snap_mmio_write64(0x1e57090, **1000**, f000000000000005)
R   hw_snap_mmio_write64(0x1e57090, **1010**, f000000000000001)
D   hw_wait_irq: Enter fd: 4 Flags: 0x10001 Expect irq: 2 Timeout: 60 sec
D   hw_wait_irq:     Got Event flags: 0 irq: 2
D   hw_wait_irq: Exit fd: 4 rc: 0
D **hw_attach_action Exit** rc: 0 Action: 0x1e57090 Base: 0xf000
D     win_size: 96 wout_size: 96 mmio_in: 28 mmio_out: 24
D **snap_action_sync_execute_job:** PASS PARAMETERS to Short Action 0 Seq: f001
R   hw_snap_mmio_write32(0x1e57090, f100, f0010100)
R   hw_snap_mmio_write32(0x1e57090, f104, 0)
.../...
R   hw_snap_mmio_write32(0x1e57090, f16c, 0)
D snap_action_start: START Action 0x10141003 Flags 10001
R   hw_snap_mmio_write32(0x1e57090, f008, 1)
R   hw_snap_mmio_write32(0x1e57090, f004, 1)
R   hw_snap_mmio_write32(0x1e57090, f000, 1)
D   hw_wait_irq: Enter fd: 4 Flags: 0x10001 Expect irq: 4 Timeout: 5000 sec
D   hw_wait_irq:     Got Event flags: 0 irq: 4
D   hw_wait_irq: Exit fd: 4 rc: 0
R   hw_snap_mmio_write32(0x1e57090, f00c, 1)
R   hw_snap_mmio_write32(0x1e57090, f008, 0)
R   hw_snap_mmio_write32(0x1e57090, f004, 0)
R   hw_snap_mmio_read32(0x1e57090, f000, 6) 0
[... 34 lines cut ...]
D   hw_wait_irq: Enter fd: 4 Flags: 0x10001 Expect irq: 4 Timeout: 5000 sec
D   hw_wait_irq:     Got Event flags: 0 irq: 4
D   hw_wait_irq: Exit fd: 4 rc: 0
R   hw_snap_mmio_write32(0x1e57090, f00c, 1)
R   hw_snap_mmio_write32(0x1e57090, f008, 0)
R   hw_snap_mmio_write32(0x1e57090, f004, 0)
R   hw_snap_mmio_read32(0x1e57090, f000, 6) 0
R   hw_snap_mmio_read32(0x1e57090, f184, 102) 0
D snap_action_sync_execute_job: RETURN RESULTS 96 bytes (24)
.../...
R   hw_snap_mmio_read32(0x1e57090, f1ec, 0) 0
D   snap_action_sync_execute_job: 23 Addr: 1ec Data: 0
D snap_detach_action Enter
D hw_detach_action Enter Action: 0x10141003 Base: f000 timeout: 60 sec Seq: 0xf004
R   hw_snap_mmio_read32(0x1e57090, f000, 4) 0
R   hw_snap_mmio_write64(0x1e57090, **1010**, 2)
R   hw_snap_mmio_read64(0x1e57090, **1008**, f002f00202000081) 0
D hw_detach_action Exit: rc: 0 CSR 0xf002f00202000081 after: 0 msec
D snap_detach_action Exit rc: 0
Step 3 took 54149446 usec
nb of occurrences = 2 
2 patterns found.
Verification of pattern number = SUCCESS!
Searching took 54149465 usec
INFO:detach response from from pslse

joergkayser, 2018-03-20:
@fhaverkamp can you comment, if the use of  snap_queue_alloc instead of typical snap_attach_action can cause the missing MMIO to 0x1000 ? Is this the "LLCMD START" ? If yes, then this missing start will confuse the hardware.

fhaverkamp, 2018-03-26:
@joergkayser The snap_queue_alloc should normally not do things differently. Since we did not implement a real queue, we just offer the interface, but what it does should be really the the same as if we use the other interface. LLCMD is done in the Linux driver, such that we do not really see this in our traces.

I think we might like to review the code anyways to see if everything is correctly implemented and the flags required to reenable/disable the irqs are correctly setup.

```
/* Context Configuration Register (CCR) */
#define SNAP_CCR        0x1000
#define SNAP_CCR_DIRECT_MODE    0x01 /* 0=Job Queue Mode
                                        1=Direct Action Access Mode */
#define SNAP_CCR_IRQ_ACTION     0x02    /* Rise IRQ when Action goes to IDLE */
#define SNAP_CCR_IRQ_ATTACH     0x04    /* Rise IRQ when Action is attached */
```

I guess if the 0x1000 is not setup with e.g. 0x4 the attach irq will not happen and if the 0x2 bit is not set the action completion irq will not occur. So that would at least explain the timeout. Question would be why the bits/register are not setup in the situation Bruno describes.

```
/* Context Status Register (CSR) */
#define SNAP_CSR        0x1008
#define SNAP_CSR_SAT    0x80           /* RO: Short Action Type for this context is assigned */
#define SNAP_CSR_ATT    0x40           /* RO: This context is attached to an action */
#define SNAP_CSR_EXEC   0x01           /* RO: Currently executing job */
#define SNAP_CSR_ACT    0x00           /* RO: Context Active */
#define SNAP_CSR_ATTACHED (SNAP_CSR_SAT | SNAP_CSR_ATT)

/* Job Command Register (JCR) */
#define SNAP_JCR        0x1010
#define SNAP_JCR_ABORT  0x0004         /* Reset Action, than Detach action from context */
#define SNAP_JCR_STOP   0x0002         /* Detach action from context */
#define SNAP_JCR_START  0x0001         /* Attach action to context */
```

bmesnet, 2018-03-29:
@joergkayser Let me know if you want me to do more tests.
Looking closer to the Trace, would the value read at address 1008 when doing the `hw_detach_action` help narrowing to the issue? Why are they different ? I'm missing how to read all bits.

> 1rst run - Step 1 : R hw_snap_mmio_read64(0x1e57090, 1008, f001f00001000080) 0
> 1rst run - Step 3 : R hw_snap_mmio_read64(0x1e57090, 1008, f002f00202000081) 0
> 2nd run - Step 1 : D snap_sync_execute_job: Error Can not attach to Action 0x10141003

Thx

joergkayser, 2018-04-16:
Please see /afs/bb/proj/fpga/framework/debug/issues_search_irun/snap/hardware/sim/ies/20180416_151015/*.log for a test with PSLSE DEBUG and application SNAP_TRACE enabled.
How cen we proceed the debug from here ?

boekholt, 2018-04-17:
The problem is in start step 3 (see Bruno's comment above):
```
R hw_snap_mmio_write64(0x1e57090, 1010, **f002**000000000001) <= ONLY 1 WRITE ????
```
Here the last sequence number is set to f002 while the current sequence number is f001 (since at that point no job is active that means that the job with sequence number f001 is the next one to be executed). So, the job manager thinks that two jobs got enqued. Therefore, the job manager does not give back the action after completion of step3, and, no interrupt for action attach is generated. But, that is what the sw is waiting for.

boekholt, 2018-04-20:
I changed the behavior of the job manager. It is now in "direct access mode" no longer taking the job sequence number into account (pull request #730).
This change is fixing this issue.
But, maybe we should review the sequence number handling of the snap library.

joergkayser, 2018-04-20:
This fixes our problem. Thanks

closed 2018-04-20 by joergkayser

bmesnet, 2018-04-20:
Great job guys. Thx