open-power/snap #686: hls_helloworld sim error on N250SP

joergkayser, 2018-03-19:
Our default text "Hello world. This is my first CAPI SNAP experience. It's real fun!" is 67B long and gets translated to uppercase fine on most cards.
On N250SP the xlation of the first 64B works, but the rest is wrong.
When trying to send a short text, the simulation fails ith a core dump

Results incl waveforms can be seen in /afs/bb/proj/fpga/framework/debug/helloworld/snap/hardware/sim/ies/20*

bmesnet, 2018-03-19:
On afs, when trying to build the model, i get an error since the dcp file doesn't exist, so I'm not able to reproduce the issue. Indeed the dcp file is not at the default path used by SNAP, but why should this file be needed when just building the model ?

boekholt, 2018-03-20:
@bmesnet The PSL DCP file is not required for simulation. But, if the environment variable `PSL_DCP` is defined, that path is checked. If you want to build a model and run simulation w/o a valid DCP you should remove (comment out) the definition of `PSL_DCP` from `snap_env.sh`. Another option would be to add the following line to `snap_env.sh`:
```
export IGNORE_CHECK_PSL="TRUE"
```
That way you will even be able to build an FPGA image using an "unsupported" PSL_DCP.

joergkayser, 2018-03-20:
@bmesnet the N250SP currently only supports DMA transfers with multiples of 128B. Is the helloworld string xferred with DMA ? If yes, we would like to postpone and disable this action, until other transfers are supported.

bmesnet, 2018-03-20:
Yes correct  the hello world string is transferred through the DMA

joergkayser, 2018-03-20:
We disable helloworld on N250SP for the moment and look at this again, when N250SP allows DMA transfers other than 128B

bmesnet, 2018-03-20:
If this helps, using a 128B aligned text may help you getting your test successful ;-)

bmesnet, 2018-09-24:
Now that DMA is final, tests ran again with different length and all executed ok.
@joergkayser Issue can be closed

joergkayser, 2018-09-27:
yes, I saw you enabled it in Jenkins, and the results are good. Thanks

closed 2018-09-27 by joergkayser