open-power/snap #715: N250SP Simulation works with snap_example_set -b11, supposed to fail

joergkayser, 2018-04-09:
When we test N250SP, we assume, that (up to now) only multiples and alignments of 128B are supported.
But sim testcases also run with snap_example_set -F/-H -b11.
Please look into /afs/bb/proj/fpga/framework/debug/n250sp-set-b11/snap/hardware/sim/ies/20180409_135509 and let us know, how this can work, and why simulation works different from actual hardware test.

ThomasFuchs, 2018-04-09:
In sim the first 11 bytes are undefined (see picture). It looks to me, that PSLSE is not able to handle such a case correctly.

[image]

joergkayser, 2018-04-09:
No, it is the other way around... Sim works, and so PSLSE may interpret 11 Bytes of 'undefined' as '0', and there is no good check in the testcase, that wrong data has been written.
@ibm-genwqe can you please check, what snap_example_set does to prove, that it has written the correct data ?

joergkayser, 2018-04-16:
New AET with PSLSE trace and SNAP_TRACE see /afs/bb/proj/fpga/framework/debug/n250sp-set-b11/snap/hardware/sim/20180416_163047

joergkayser, 2018-04-18:
The state 'undefined' was interpreted by PSLSE as 0xFF, which was the default 'uninitialized' pattern in the testcase. After changing the default pattern to 0xE5, the testcase fails as expected, and we should have a somewhat better test coverage in the future.

closed 2018-04-18 by joergkayser