open-power/snap #726: cloud doesnt open GUI with 'make snap_config'

joergkayser, 2018-04-18:
When following the README guidelines in "SNAP-DESKTOP V0.1.1", the step 'make snap_config' fails to open the GUI, because $SIMULATOR is not defined.
This can be circumvented by changing snap_env line 151 from $SIMULATOR to "$SIMULATOR" to prevent having a unary operator.

boekholt, 2018-04-18:
Fix is in pull request #727

joergkayser, 2018-04-20:
This works in our 'cloud regression'. @kenhill  Will you pick this as well ?

joergkayser, 2018-04-25:
waiting for the next $CLOUD_BRANCH, this is still 1.4.1 and does not contain a fix yet

joergkayser, 2018-06-07:
The SNAP_DESKTOP v0.1.1 still uses $CLOUD_BRANCH=1.4.1, and snap_env line 151 fails and the GUI cannot be opened, see above description.
Please fix as described above. Thanks

kenhill, 2018-06-14:
@joergkayser, sorry for the delay. SNAP_DESKTOP v1.0.2 uses SNAP v1.4.2.

Is this still a problem?

joergkayser, 2018-06-18:
@kenhill SNAP v1.4.2 incorporates the fix #727 from @boekholt. When I call https://platform.jarvice.com, then I see SNAP_DESKTOP v0.1.1(408985), which sets the environment variable SNAP_DEV_RELEASE=V0.1.1
How can I upgrade SNAP_DESKTOP from v0.1.1 to v1.0.2 ?

kenhill, 2018-06-18:
@joergkayser not sure why see the older version. Can you try the new portal: https://mc.jarvice.com

The apps should be public under the `Compute` tab.

joergkayser, 2018-06-22:
The todays Login finally shows SNAP_DESKTOP v1.0.2 and uses CLOUD_BRANCH v1.4.2, which solves this problem. Thanks

closed 2018-06-22 by joergkayser